utkabobr/DuckPro-Kobra3 #1: Missing moonraker_url in config.json causes Mainsail to get stuck initializing.

bearclaw92, 2024-10-05:
After successfully flashing the Duck Pro CFW on my Anycubic Kobra 3, I encountered an issue where Mainsail would get stuck on the "Initializing" screen. Upon investigating, I discovered that the config.json file in the Mainsail directory (/useremain/home/ytka/mainsail/config.json) was missing the moonraker_url setting.

Manually adding the following line to the config.json file resolved the issue:

"moonraker_url": "ws://<printer_ip_address>:7125",

Suggested Solution:

Update the Duck Pro CFW update script to ensure the moonraker_url setting is correctly added to the config.json file during installation.

Temporary Workaround:

Manually add the moonraker_url setting to the config.json file after flashing the firmware.

utkabobr, 2024-10-05:
Thanks for your investigation!

Sadly, config.json can't be modified by update script because IP is not a static thing and can be changed at any time by router's DHCP settings (For example, your printer's IP was 192.168.1.100, but after a printer or router reboot it's now 192.168.1.125, that is a default configuration on some routers)

Default config.json should work fine with auto-discovery. Have you installed any privacy extensions for your browser that can mess with hostname?

bearclaw92, 2024-10-05:
okay yes I understand now, I am still not sure why I had that issue. Thank you for your response. I had the issue across several browsers, maybe its something I have set up and am not aware of. I'm just glad to have mainsail, thanks a ton this firmware is great.

closed 2024-10-05 by bearclaw92