utkabobr/DuckPro-Kobra3 #2: Bricked the printer

softwaresupportster, 2024-10-05:
followed the instructions and now the printer is bricked!!!

EnderPy, 2024-10-05:
I did aswell. I had some error where it was asking to check hotend cables and stuff
I followed this https://github.com/systemik/Kobra3-Firmware/tree/main/Disaster%20Revovery
To reflash to stock firmware and it fixed it

utkabobr, 2024-10-05:
The software is provided at your own risk. If you can still access your ADB/SSH, then you can use it to restore start.sh script to stock one (Attached in comments, no modifications made, so you WILL lose access to ADB/SSH after this, please check other files beforehand), this should fix errors on the screen. Also, you can revert to the stock firmware with stock swu file as described in comment above.
[start.zip](https://github.com/user-attachments/files/17266462/start.zip)

utkabobr, 2024-10-05:
There was also a bug that created wrong printer_data folder (Fixed if moved manually or flashed newer build), should be fixed in a new release (https://github.com/utkabobr/DuckPro-Kobra3/releases/tag/05.10.24)

softwaresupportster, 2024-10-05:
Unable to fix it now it’s stuck on Anycubic screen with froze. Loading bar

utkabobr, 2024-10-05:
There is a way to connect through UART if any other way doesn't work. Baud rate is 115200, it provides direct access to console.

[image]
Image from here: https://t.me/K_3_D/1944033/3078681

softwaresupportster, 2024-10-05:
Amazing, thanks il look into

softwaresupportster, 2024-10-06:
is there +5v on the other header or is that not needed

utkabobr, 2024-10-06:
It's not required, motherboard should be powered externally (not from UART)

softwaresupportster, 2024-10-06:
managed to get a response but no console

[SERIAL/DIRECT] CONNECTED TO PORT COM8 (115200-8N1)

DDR V1.11 285bced5c1 wesley.yao 23/03/28-14:23:36
S5P1
f967
rgef1
DDRConf2
DDR3, BW=16 Col=10 Bk=8 CS0 Row=14 CS=1 Die BW=16 Size=256MB
924MHz
DDR bin out

utkabobr, 2024-10-06:
Check if it responds to the keyboard. If not, check that you don't have loose wiring

softwaresupportster, 2024-10-06:
checked everything, tried different usb-ttl and still the same its well and truly bricked, can it be flashed? with rockchip tool

utkabobr, 2024-10-06:
I am not sure about how rockchip tool works. If it accepts raw image files, then you can try to get them from files/update_ota.tar and flash via it

softwaresupportster, 2024-10-06:
[image]

Also is this marked connector a programming port

Got new mb on order so will be backup and running soon but it would be great to have a solution for a bricked board

weekendkoder, 2024-10-07:
@EnderPy

Maybe this Backup/Restore Solution can Help you to "unbrick" your board:
https://github.com/Bushmills/Anycubic-Kobra-3-rooted/discussions/5

EnderPy, 2024-10-07:
@weekendkoder 
my Printer is fine, like I said I fixed it by flashing stock firmware over USB and it fixed it

softwaresupportster, 2024-10-07:
That is brilliant thank you at least now we have a solution il post the results

weekendkoder, 2024-10-07:
I see :-) . Somehow my message with the Backup/Restore should be directed to @softwaresupportster. I guess, i was too late on my computer again. Glad the information is useful to some users.

Rand0mcookie, 2024-10-16:
> That is brilliant thank you at least now we have a solution il post the results

Did you ever find a solution, I seem to be in the same situation

urnero, 2024-11-21:
Hello, I have the same 11407 screen. I have connected via adb, and the IP in browser shows some connection failure to the mainsail. How can I restore my start.sh script?

KungPhoo, 2024-12-01:
I inserted the update USB stick, waited. The old firmware was running. I saw no sign of an update. I heard a music playing. So I shut the device off and on after a few minutes.
Now I have an 10131 error screen. Downgrading to any older firmware doesn not seem to work either. What can I do?

weekendkoder, 2024-12-03:
@KungPhoo ... thank you, for the tip. It helped me. It saved my Printer !