utkabobr/DuckPro-Kobra3 #3: Can´t install the update.

Spidywert, 2024-10-05:
Hello,

I have a few issues, first of all i must rename the Folder also to Update on the latest Firmware (2.3.3.9) it didn´t update the Software on the fly.

And a secont one are a Message of that the Local Storage are insufficante and i have to remove some files, but the Storage was allredy empty.

Thanks for that Work.

utkabobr, 2024-10-05:
So it doesn't react at all if you don't rename a folder and just boot with it?
Newer versions of anycubic's firmware should try to install it automatically, once install is started you should hear "At the doom's gate" tune.
Do not reboot printer at least 15-20 minutes to ensure everything is installed properly.

Local storage might mean internal memory of the motherboard, please check how many space does your gcodes consume.

Spidywert, 2024-10-05:
after doing the other firmware i got the a reaction. and a while there was a connection to klipper. but i will try it again.
 I have deletet all the unnecessery files from the local storage.

Spidywert, 2024-10-05:
soo, i have it working. try it to stick the usb stick after the rebooting and waitet for je music. it seems like it was my fault 👯. thank you for Help

utkabobr, 2024-10-06:
Closing as a successfull installation

closed 2024-10-06 by utkabobr