utkabobr/DuckPro-Kobra3 #4: general question about "Printer.CFG"

weekendkoder, 2024-10-07:
Hi.

I am new to Klipper. Just to verify ... is it possible to modify the printer.cfg to my own needs ? As i understand the "printer.cfg" is the main important file, that holds all parameters that are necessary for Klipper. Is that right ?

If we are able to modify the "printer.cfg", do we need to flash the firmware always, when we want to do small changes to something?

bearclaw92, 2024-10-07:
https://www.klipper3d.org/Overview.html

utkabobr, 2024-10-07:
No, it's not required to reflash firmware. You can edit that file and click "Save & restart".
But you should note that it's not a regular Klipper, it's goklipper - anycubic's heavily modified fork on golang, it could be missing some important features that you want to enable in printer.cfg

utkabobr, 2024-10-07:
Yup, you should get your IP from device's screen. All you need from documentation is https://www.klipper3d.org/Config_Reference.html
It's all about printer.cfg, but Kobra-s on klipper are running goklipper, it may behave different in certain ways

weekendkoder, 2024-10-07:
Okay. Thanks. Now it is clear to me.

weekendkoder, 2024-10-07:
I forgot to ask an other question which is related to printer.cfg. If we want to do changes, we need to unpack "update.swu". Therefore we need a password to unpack it. Do you have it ? And afterwards, if we did our changes, we need to zip the folder again to the file "update.swu". But while repacking, it need to be zipped with the same password, i guess?

bearclaw92, 2024-10-07:
@weekendkoder no you don't need to do any of that. To make changes in printer.cfg just edit via mainsail and then click save and restart. I recommend you make a backup before editing.

weekendkoder, 2024-10-08:
Got it. Somehow i missed it, that MainSail is able to do instant editing of files, like "printer.cfg". Yes, i will do backups, before i do some big changes. Thank you for the reminder.