utkabobr/DuckPro-Kobra3 #6: Kobra ACE Pro - FW update

weekendkoder, 2024-10-07:
General Question is: do we need a Kobra ACE pro FW - Update too? Or does the "DuckPro-Kobra3"-update-package contains also a FW-Update for it ?

utkabobr, 2024-10-08:
ACE Update installs separately, so you can update it from the UI

weekendkoder, 2024-10-08:
Okay. Got it. Thank you.

OoohPossum, 2024-10-09:
Hello @utkabobr ,

do you plan to integrate a potential reverse-engineered access to the ACE Pro engine
(potentially based on this undertaking: https://github.com/printers-for-people/ACEResearch ?)
.. into your firmware for a fully independent usability of the Kobra 3 ACE-bundle?

Or is it already possible to use the filament-changing functions of the ACE pro connected to a Cobra 3 printer with the DuckPro firmware? From how I read the description, that's not yet a given?

Thanks in advance, as well for the great alternative firmware! :-)
Very happy that the Kobra 3 gets attention by open source coders.
AnyCubic does unfortunately not seem to keep "we're making it open!"-promises, ever.

weekendkoder, 2024-10-12:
As we have the the Kobra 3 main FW Update-file, named f.e. like "K3_v2.3.3.9_FW-update.swu" (basicly a min. 100mb *.swu file), It would be great i we also would have a "ACE pro" firmware update file. I don´t know where @utkabobr and other repositories got a *.swu file. I could not find any *.swu files on the original Anycubic-site. At the link where it should be, it is empty.

I was forced to connect my printer with the anycubic-servers. This is how i got the only chance to update my ACE pro. From vers. 1.3.73 to 1.3.83.

I think it would be nice, if we would have the Kobra 3 and ACE pro FW "...update.swu"-files somewhere stored, maybe on gitHub.
Does anybody know if there is a secret link to these *.swu files (from the anycubic-servers)? I would like to have the "ACE pro v1.3.83"-update.swu locally stored on my archive.

utkabobr, 2024-10-12:
I've got it from the update logs

utkabobr, 2024-10-12:
Miss clicked from phone :(

weekendkoder, 2024-10-12:
Cool. It would be great, when you find some time, to post the log´s here. Somehow i like to have these update-*.swu files also for the ACE pro. Just to keep them on some drive, or gitHub.

weekendkoder, 2024-10-15:
Perfect ! ... Thank you.

Jookia, 2024-10-23:
If someone's willing to perform an ACE software update with some test tools enabled, please check out https://github.com/printers-for-people/ACEResearch/issues/4 or contact me directly. I have ACE_1.3.82 from above and ACE_V1.3.76 locally downloaded from a random Facebook group, but I'm a bit afraid of posting it online :) This process would not be destructive.

utkabobr, 2024-11-05:
Uploaded 1.3.84 too: https://ytkab0bp.ru/ACE_1.3.84.swu

OoohPossum, 2024-11-07:
@Jookia : Sure. As long as I don't have to upgrade to the new printer firmware 2.3.5.3, I can do that.

But my printer is not "online", it's LAN only, printed via modified Orca Slicer profiles and mainsail via DuckPro (based on 2.3.3.9).

If I can help you on the way of reverse engineering the ACE Pro communication, which is a great project and will help probably a lot of users with other printers (or even the Kobra 3 with custom firmware) to hybridize their printer with an "open source" access to the ACE pro, I'll do that.

If you communicate which changes on the printer (based on the actual DuckPro 05-10-24.02 might be necessary and which log files / log levels you need, I'll provide those.

Jookia, 2024-11-07:
> @Jookia : Sure. As long as I don't have to upgrade to the new printer firmware 2.3.5.3, I can do that.
>
> But my printer is not "online", it's LAN only, printed via modified Orca Slicer profiles and [...]

Hi there,

Would you mind emailing me at contact@jookia.org so we can discuss this further? I don't want to clog up this issue with step by step instructions, and I can't guarantee the logs won't contain sensitive data of some kind.

Jookia.

OoohPossum, 2024-11-07:
@Jookia : Will do that.