utkabobr/DuckPro-Kobra3 #8: Install OK But ...

pyrah, 2024-10-08:
Hi, 

The installation is fine, but I’m stuck when trying to call the machine’s IP and access the web interface. Any idea what I can do?

[image]

[image]

Thx ;)

pyrah, 2024-10-08:
Finaly after 3 try i have this : 

[image]

I think is ok ? :)

utkabobr, 2024-10-09:
@weekendkoder definitely not required. My Kobra 3 haven't connected to any Anycubic account ever, it prints via moonraker (from PrusaSlicer) just fine

weekendkoder, 2024-10-09:
Ah. Thats nice to hear. I was afraid that we could have been forced by anycubic to only connect to the own printer over their servers. Glad its not the case :-)

MrSolidus89, 2024-10-17:
by installing this FW, can I continue to use anycubic services? for example its remote app?

Nexa23, 2024-10-22:
dear friend, tell me how to add a web camera?

utkabobr, 2024-10-31:
> by installing this FW, can I continue to use anycubic services? for example its remote app?

Should be fine if Anycubic doesn't involve some new changes, I haven't disabled any of their services. But please don't mess with device.ini file or device keys, they're required to work with cloud