utkabobr/DuckPro-Kobra3 #9: DuckPro-Kobra3 FW update - success review

weekendkoder, 2024-10-12:
Hey all.

I got the DuckPro-Kobra3 FW update (rev. 05.10.24-02) and it worked out of the box flawlessly. I was on an very old FW vers. 2.3.1.3_1.1.6. In the beginning i was **not able** to update the "DuckPro-Kobra3 FW" via USB. I always got the error message: "Insufficient local storage capcaity, please delete the files". First, i did not know what really the matter was. Because i had free space of min. 2GB, out of 5GB i think. I got confused, because i believed 2GB is enough capacity for a small FW update !? Nevertheless it turned out, that i needed to delete all my print *.gcode files from the local storage. I the end i **freed up 4gb space**, i think. After this, the "DuckPro-Kobra3 FW" via USB worked. Just wanted to let other people know, if they maybe run into the same issue.

I always got problems with my leviQ abl, to accomplish a perfect 1st layer. After the update with "DuckPro-Kobra3 FW" it seems i dont have these issues anymore. Need more time to evaluate, but for now it looks like, the new FW do a better job. Second great thing is access to "MainSail". I did not explore so much inside, but for now it is great that i can see for example my Bed-Leveling-Visualization. We are also able to add own-macros as i saw. It would be great if we could have some kind of camera function. Lets see. I am very happy, with this release. Thanks for doing this for all kobra3 users.

systemik, 2024-10-14:
Maybe the git-core folder needs to be remove from the distribution. This take a huge space and is surely slowing down update (and I imagine that people who broke their printers did shutdown while the update thinking it was over)

weekendkoder, 2024-10-14:
Okay. Thanks for the hint. Hm, i guess "git-core" is not needed, for an OS-update. Maybe we can remove it manually (unpack, delete, repack), or the owner of this repo (utkabobr), will decide to remove it or leave it inside. He should know if such a folder is necessary. I pointed out, that i have had just an issue because of lack of enough space. I think the "official FW" which is packed, is about 155mb big in size.

utkabobr, 2024-10-31:
> Okay. Thanks for the hint. Hm, i guess "git-core" is not needed, for an OS-update. Maybe we can remove it manually (unpack, delete, repack), or the owner of this repo (utkabobr), will decide to [...]

You can clone repo and delete it there without re-packing already built image, it's just longer. Use build script after modifications

weekendkoder, 2024-12-02:
Okay....good to know. Thank you.