utkabobr/DuckPro-Kobra3 #11: Ace problem?

sebb3, 2024-10-18:
Every time a start a print from mainsail or orcaslicer, I get

error: typ = WebRequestError, code = 10011703, message = unknown filament in extruder

First I thought there was some stuck filament in the printhead, but the problem doesnt occur when starting the print from the touchscreen.

Is this because it doesnt know which filament slot to push, and is there a way to configure it through some custom g code?

Basically, since this happens every time it makes this whole concept pointless, so I suppose others dont have this issue?

utkabobr, 2024-10-18:
Just add `T{initial_extruder}` to your start G-Code. Filament change: `T{next_extruder}`

Felixoid, 2025-01-13:
Hey-hey. I just installed the firmware, and it doesn't print with ACE connected.

Neither `extrude` in the web interface works.

I feel, the g-code isn't full. Could you give a full necessary command, please?

closed 2025-04-08 by sebb3