utkabobr/DuckPro-Kobra3 #20: Can you add camera control from another custom firmware? link included

joelpr1234, 2024-10-27:
https://github.com/systemik/Kobra3-Firmware

The above firmware includes a camera control, streaming and light control, if you could add this to DuckPro-Kobra3 firmware will be great. Unfortunately I don't know how to do it. This can be linked to Mainsail and have the streaming available.

Thanks, I have 2 K3 with this firmware.

systemik, 2024-10-27:
This has been changed today on my side (repo updated). 
There is a way by using mjpg-streamer to configure the camera in mainsail (view only). Maybe repo owner can add this to his repo.

kwe1009, 2024-10-30:
> > This has been changed today on my side (repo updated). There is a way by using mjpg-streamer to configure the camera in mainsail (view only). Maybe repo owner can add this to his repo.
> 
> I was looking what you've done on your repo to add the camera to mainsail. I'm lost. I have no idea how i can add the camera the way you've done it. Is it possible for you to write a small tutorial to describe how we can have the camera working? Thanks a lot.

https://www.youtube.com/watch?v=-Z47-tel_8A

utkabobr, 2024-10-31:
> This has been changed today on my side (repo updated). There is a way by using mjpg-streamer to configure the camera in mainsail (view only). Maybe repo owner can add this to his repo.

I can configure it to work on autostart if you don't mind. And I think it will be better for you to create a pull request with mjpeg streamer included in binaries, so you'll be displayed as a contributor

weekendkoder, 2024-10-31:
@kwe1009 Thank you, for the tip. This camera hardware seems to be really cheap. And like in the Video described working autonomously, because over ip / usb. Maybe i buy 2x ...