utkabobr/DuckPro-Kobra3 #21: After successful installation running into a bug with klipper start-up by configuration change (or restart?).

OoohPossum, 2024-10-27:
Hello,

my setup is a Kobra 3 Combo with ACE pro. It had firmware 2.3.1.X (9?) installed on delivery.

First installation procedure of DuckPro 5.10.24-2 failed, it led to error 11407, even after finished installation (update.swu got removed from the USB drive).

I then updated to 2.3.2.9 original firmware, which worked fine. After a second installation of DuckPro 5.10.24-2 it booted successfully and was usable.

Bugs noted:
- Setting fan speed on the display didn't work anymore (no activation possible with the switch, always 0% speed, not configurable).

After a setup of Orca Slicer and configuration of the WLAN connection to an offline access point the usage of Mainsail with moonraker worked fine: the ACE pro behaved well, if printing was started over the display (after upload of sliced gcode file through mainsail).

Didn't try out starting printing process over mainsail directly, yet - in regards to selecting the desired filament spool of the ACE pro before start of the printing process (was easier over the display than with a modified gcode / command).

The problematic bug occurred after I configured mainsail and the printer.cfg.
After I changed some (not critical, like minimum temperature for extruder movement, mesh grid 10x10 / 9x9 instead 5x5) settings and used mainsail web interface to "save and reboot", the machine hang on boot at error 11407, again.

The configuration of the printer itself (WLAN connection, mainsail service, moonraker service) still worked - but it couldn't connect to the klipper service anymore.

I then tried a lot of possible ways to overwrite the whole configuration:

DuckPro over DuckPro directly => worked, but configuration was untouched, as such same problem.
Original firmware 2.3.2.9 over DuckPro => worked, booted machine, but surely enough without mainsail/moonraker/modified klipper service

DuckPro again over OFW => worked, but same issue (error 11407) and no connection to klipper service - but mainsail connected automatically (as such configuration persisted).

Then removed the mainsail-accessible configuration file folder, flashed OFW again, machine booted.
My hope was that if the configuration files are no longer existent in the mainsail-accessible folder (didn't use SSH yet), it would replace those by re-flashing of DuckPro with the standard configuration files. But same error again: 11407.

I then flashed the OFW, reset the printer wholly to factory defaults over the accessible display option. After a reboot I set up the OFW by display directly for "first usage". This all worked without a flaw.

My idea was that after factory reset all configuration files would cease to exist and after a new flashing of DuckPro (after the "first usage setup") it would start-up without hassle.

But no, still error 11407.

If this is linked purely to the printer.cfg (or its folder), then I assume that "editing / saving it by mainsail" saves or moves it to a wrong folder? Or it (or another configuration file) got corrupted by the editing procedure.

If it is linked to a bug on "save and reboot" at mainsail web UI, that would be weird, but how to fix it, as long as the firmware leaves "old" configuration files on the machine at re-flashing.

Quite confusing that it still occurs after factory reset and re-flashing. The original firmware boots without hassle after each flashing procedure, tho.

Not sure if it is linked to the problem: The latest DuckPro re-flashing procedure wasn't left alone for 20 minutes, but I as well don't hear the beeper tune (hearing impaired). That could play a role, I had no time to wait a full 20 minutes at the last try after factory reset.

The OFW (1/4th the size) was flashed in under 2 minutes and re-booted on itself, while the update.swu file is deleted from the USB storage after flashing procedure ends, giving a clear signal about finished procedure (on each re-flash).

I would recommend that the DuckPro firmware does the same: automatic reboot and it's clear that it is done.

It should be possible for me to setup the network through the working OFW(!) display configuration, then flash DuckPro over it (even waiting 25 minutes, but I'm not at the company now, as such has to wait a day or two) and then connect to mainsail without klipper service access, again (even after factory reset), even with error 11407 occurring.

Or access the machine after this procedure over SSH, if there's an idea where the problem occurs at: folder structure, file location, .. I could modify it manually (had no time yesterday).

derrene666, 2024-10-28:
same issue here. friend of mine got the same problem!

kiinami, 2024-10-28:
I had more or less the same problem when modifying configuration. The save function of Mainsail works perfectly for me, but sometimes modifying innocuous things leads to goklipper crashing on start. I think part of it has to do with the fact that goklipper does not use exactly the same config schema for printer.cfg, resulting in goklipper not booting up when something is wrong in `printer.cfg`. I have been fixing it through SSH every time it does not work (route is `/useremain/home/ytka/printer_data/config/printer.cfg`). The error can be seen doing `cat /tmp/gklib.log`. IDK if that helps.

utkabobr, 2024-10-31:
> I had more or less the same problem when modifying configuration. The save function of Mainsail works perfectly for me, but sometimes modifying innocuous things leads to goklipper crashing on start. [...]

Please note that goklipper's logs is much useless by default. You should modify /userdata/app/gk/config/api.cfg, replace `"logLevel":"info"` with `"logLevel":"debug"`. Then restart gklib or entire system

derrene666, 2024-11-01:
I solved it for myself, but I think it was purely by chance. After 4 hours and ~5 minutes it now worked!

I don't know if the information is of any use, but the K3 was a pre-order so one of the first

OoohPossum, 2024-11-03:
At company and fixed the issue.

Some aspects which I verified for my system and DuckPro 5.10.24-2 as well as for the original FW:

- Neither DuckPro nor the original firmware will replace lost, deleted or incompatibly modified files in folders which are not linked to basic core functionalities of the OFW.

- It is not possible over the automated update.swu flash routine to downgrade (easily) or "re-flash" the same firmware again with replacing of the already once flashed files in some of the folders at least (didn't try out all of them).

- This means that even if you flash OFW 2.3.2.9 over OFW or DuckPro based on 2.3.3.9, it will not replace files or folder structures like the configuration folder (probably the whole ytka-folder, at least).

- To reset the device to factory configuration will as well not fix errors in relevant files which are linked to any CFW, if those files are not relevant and basic to the OFW as well.

- But it replaces some configuration files which makes it possible to **boot** OFW successfully, set up WiFi connection, then re-flash DuckPro (wait for a long time, better 40 minutes on some systems or configurations) and then reboot it. It did **never** reboot automatically while flashing DuckPro, while it does so all the times with OFW, even if the flashed one is a downgrade (lower version number).

- After WiFi is set up and DuckPro is running into error 11407 on touchscreen again, if configuration files are missing or defective (unusable parameters/values, doesn't matter, it will always run into this error on the screen, even if moonraker service can't start), you can connect to the mainsail web UI again, but either klipper or klipper & moonraker service will not be started. As such you can only access the one log and configuration folder.

- You can't simply overwrite a defective configuration file with a backup file in mainsail web UI. Only way to do so is to open a backup / intact version in the editor (if available in the accessible config folder over "machine" tab), copy all of it, then close the editor, open the real config file and replace all of it's content, then save it.

- Better option is to use SSH or ADB access to replace the files via either upload from your computer or copying of a backup over the defective real configuration file. Should be self-explanatory, but the process can be tricky, but it is well possible.

- I tried to access a backup configuration file on USB memory storage over the mainsail editor function and paste it into the real configuration file, but the access to the usb device with ability of opening a configuration file in the editor doesn't seem to work without quite a lot of changes, unfortunately.

- **Easier**: to create a backup of all configuration files right in the configuration folder **before** you change them manually. Then you can connect via SSH to it, even if klipper / moonraker service fails to start after changes and copy it back by the ssh console without transferring anything from your computer or USB memory stick to the printer beforehand.

- **Even easier**: Create the backup and change one aspect in the editor (mainsail), save it and restart the machine. Best would be: manually to check if touchscreen runs into error 11407. The klipper service startup failure in moonraker as well is a good indicator of a non-working change.

- **Warning**: mainsail web UI editor "save" function is definitely bugged! It works more reliable with "save and reboot", but I as well saved it successfully by clicking two to three times "save". If I only clicked it once and rebooted the machine, in around 80% of my tries it did **NOT** save the changes and reverted back to the unsaved previous version! This is very confusing, but if one knows about it, it is preventable.

- If boot-up of the services does not work due to configuration changes, it is the easiest way to revert that **ONE** change you tried out (and changed) last! Then save & reboot or multiple clicks on save, then reboot machine. It will work again, even without having to need SSH to replace the whole file, if it got saved successfully. The editor itself works fine, but the save function seems bugged, why ever.

So far my analysis.. Got it running, changed some aspects of the printer.cfg, tried out which setting directly results in a failing klipper service start:

Mesh grid from 5,5 to anything above 6,6 will **NOT** work. I tried all combinations, anything above 6x6 points ends in not starting klipper service.

As such: "my" issue with it is solved, but if there _could_ be some modifications to mainsail web UI, which allow for a direct opening of the editor to any path accessible (as well to plugged in USB storage media) and/or a sort of file copy/move command file manager accessible by it, that would surely help those which have problems handling SSH connection or ADB file transfer.

Maybe the touchscreen menus allow for a direct file transfer from USB storage media to the internal storage folders, I didn't try this out with the configuration files.

Remember: Don't delete configuration files or the custom folder structures, instead create a backup prior to modifying them, as else you might have additional steps to solve the problem of services not starting manually. Re-flashing OFW or DuckPro does not work in this case, it will simply leave an emptied configuration folder empty (I tried that out to verify it).

I seriously hope that no one uses SSH to eradicate important system files which are necessary for SSH / ADB access. Because there's a chance that a re-flashing will not replace them, even if the files are existent in the compressed firmware package.

closed 2024-11-03 by OoohPossum

derrene666, 2024-11-03:
Mesh 7x7 and above work! you have to change the algorithm to "bicubic"

[image]

OoohPossum, 2024-11-03:
Hello @derrene666 ,

thank you for the information! I actually set it to bicubic before the try-out of the mesh grid, but that was before I noticed the save-bug!

It is well possible that it didn't get saved the first time and after I noticed the failed saving feature, I didn't focus on or checked the algorithm anymore. Have to re-do the tests with 7x7 and 9x9 - and watch the algorithm for faultless saving, beforehand.

Thanks again, I wonder how often the save-bug hit my tests and configuration before I noticed it, actually. Dang..

OoohPossum, 2024-11-04:
Tried it out today, proceeded the "saving" multiple times until algorithm was persistently saved, now 7x7 works fine.

If there are settings which require other configuration changes as prerequisites, I now set the prerequisites first, save it, restart the machine and check if it was saved successfully.

Reduces starting errors of klipper service based on partly incompatible configuration by a lot.

I wonder why the "save"-function in mainsail web UI doesn't seem to work as reliable as the "save and restart" function. Even that one doesn't work 100% for my setup. I had the case where I saved & restarted, switched the printer off and let it boot again (to check the touchscreen changes) - and saw that the changed settings were reset to the prior-save state.