utkabobr/DuckPro-Kobra3 #22: Stock Firmware

OXOTH1K, 2024-11-01:
Приветствую! Как я могу вернуть дефолт прошивку на моем принтере? 

Greetings! How can I return the default firmware on my printer?

utkabobr, 2024-11-01:
Ссылка на стоковый файл лежит в README, нужно переименовать его в update.swu и следовать инструкциям по установке

OXOTH1K, 2024-11-01:
> Ссылка на стоковый файл лежит в README, нужно переименовать его в update.swu и следовать инструкциям по установке

Первым делом так и попробовал. Скинул его в папочку с символами на флешку, вставил, подождал час, перезагрузился - ничего не поменялось. Попробовал сбросить в дефолт - ничего не поменялось

OXOTH1K, 2024-11-01:
Может потому что выложена столовая 2.3.2.9 а на принтере стоит 2.3.3.9?