utkabobr/DuckPro-Kobra3 #23: Moonraker Timelapse

pedro2d10, 2024-11-02:
I'm struggling installing [moonraker-timelapse](https://github.com/mainsail-crew/moonraker-timelapse). Each time I add the macros in the printer.cfg file, Goklipper crash.

Someone know where I can find the logs for goklipper?

kiinami, 2024-11-02:
Since goklipper is written in Go, all the code inside the macro has to be written in Go too, following the specification of the template engine used, which is [pongo2](https://github.com/flosch/pongo2), whereas normally it would be written in Python using the Django template engine. 

Looking at the `timelapse.cfg` file that needs to be added, it looks like it is going to be quite a ride to translate everything to Go, but if you do end up making it happen, please do share. Good luck!

pedro2d10, 2024-11-02:
Thanks for the details. I was going to abandon the idea. But now I now why it does'nt work. I'll try to translate it but I'm quite a noob in go.