utkabobr/DuckPro-Kobra3 #26: Firmware 2.3.5.3

wacksparrow, 2024-11-05:
Will I be able to install the new stock firmware "over top" of this or should I reset to factory before trying the new firmware?

utkabobr, 2024-11-05:
I think it should be fine, but you will lose any DuckPro's features

wacksparrow, 2024-11-05:
I know it's early still but do you know if duckpro is compatible with the new firmware? I hear the new firmware corrected the z offset issues that's been plaguing me since new

utkabobr, 2024-11-05:
I can just backport goklipper from the new firmware if it's the case

wacksparrow, 2024-11-05:
You the man. I really appreciate it.  Will be watching for updates. Thanks for everything

qwurete, 2024-11-05:
[image]
When I inserted a flash drive with custom firmware, a beep sounded and the printer began loading, but the loading bar did not fill. After that I wanted to go back to stock firmware. but still the same problem, now the printer does not respond to anything, tell me how to solve the problem. Firmware version 2.3.5.3

wacksparrow, 2024-11-05:
I can confirm 2.3.5.3 installed over top of DuckPro (from the cloud with install initiated from the Anycubic Slicer Next desktop app) with no issue, no need for factory reset and no need to uninstall DuckPro, also, as expected I did lose DuckPro features. Changes I've noticed so far are the Camera LED now works and the camera is accessible in the Anycubic android app while I'm on different networks and/or cellular. Z offset seems to be better now too but more testing is needed to confirm stability.

nighty001, 2024-11-05:
Hi, I also went to 2.3.5,
unfortunately there is now no possibility to go back to the Duck.
It beeps once and after 30 minutes nothing has happened, it remains 2.3.5.

The Duck ran better except for
- no display on the printer when printing is active via LAN
- via OrcaNext from Anycubic it freezes during the transmission of the printout 

Is there a way to adapt the Duck to the new software so that it can be reinstalled?

Translated with DeepL.com (free version)

OoohPossum, 2024-11-06:
> Changes I've noticed so far are the Camera LED now works and the camera is accessible in the Anycubic android app while I'm on different networks and/or cellular. Z offset seems to be better now too but more testing is needed to confirm stability.

The AnyCubic newsletter regarding the update agrees with you in terms of the camera light switch, it states:

> Kobra 3 Printer Firmware Update:
> 
> 1.Camera LED Light Switch: A new independent toggle switch has been added to the firmware UI and the Anycubic app, allowing you to turn the camera LED light on/off. This feature is only available for cameras purchased directly from Anycubic.
> 
> 2.M601 Command Support: The firmware now supports the M601 command, which allows the printer to automatically pause when the specified height is reached, as indicated by the command in your slicing software.
> 
> 3.Pre-Heat During Remote Task Sending: When sending a task remotely, the printer will preheat during the file download, reducing the startup time for printing.
> 
> 4.Hotend Cooling During Pause: During print pauses, the hotend temperature will automatically decrease to 140°C, reducing the risk of filament leakage and preventing clogging due to prolonged high temperatures.
> 
> 5.Optimized Video Function: The video playback smoothness has been improved for a better user experience.
> 
> 6.Enhanced Print Protection: Additional print protection features have been added to safeguard the printer in more unusual scenarios.
> 
> 
> ACE Pro Printer Firmware Update:
> 
> 1.Improved Filament Entry: The smoothness of filament entry has been enhanced for better overall printing performance.

Most interesting aspect of this is M601 support for me. It basically allows to pause printing on overhanging structures without having to print supports, as it will pause automatically and I can move the extruder manually via mainsail to the desired support / overhanging position, filling it by manual extruding with a "spool" of filament, which acts as low-filament waste replacement for a support stucture.

Then "resume" and it prints just fine. A bit work, tho.

Also allows for printing of constructions including hollow structures, where it stops before placing the first top layer, which allows one to insert the fitting reinforcing / mounting parts, like hexagonal nuts which get fully integrated and encased in this way.

Could be that the DuckPro firmware already supported M601, I didn't test it yet on this printer. But official support is always a good improvement.

I wonder what they did improve on "Enhanced print protection". If it is related to the mesh recalibrating Z offset variation which occurred twice to me (resulting in hardened nozzle vs. spring steel bed - hooray), a good thing as well.

Peschi90, 2024-11-06:
Im pretty confused, a lot guys at Facebook report better bl and z level. Why should Anycubic not use this in their newsletters or change logs? It's one of the biggest points against the printer and they didn't tell the world that they improved it?! 

Sounds crazy, I will wait till it's confirmed that they improved it.

Peschi90, 2024-11-06:
Did you factory reset after update to newest version?
This is important!

nighty001, 2024-11-06:
Yes, even downgraded again to get back to the Duck, which unfortunately didn't work. Then back to the new one as the print head can no longer be downgraded ...
Then switch back to the new one and reset.
It then prints well up to 3-4 prints.
If it is leveled before each print and the nozzle is cleaned with a brush, it works reasonably well.

nighty001, 2024-11-06:
In addition, I have equalized my bed with foil to an average value of 0.098 mm, previously 0.8 mm deviation. 
No comparison to the Duck with activated default mesh in gcode

OoohPossum, 2024-11-07:
**Downgrading or "same version"-reflashing doesn't fully work via USB memory sticks plugged in**, at least not without (unknown to me) changes to the OFW.

I tested this with the firmware the actual DuckPro is based on, a previous OFW and an older one (from my printer), it keeps most of the configuration and doesn't seem to overwrite all files (on same version doesn't restore all of them, as well).

If something goes faulty by modifying files which don't get replaced via re-flashing, one has to replace those manually (SSH / ADB access, if still working - or direct access to the (on newer boards removed.. AnyCubic does it again..) USB molex connector, which is placed directly on the PCB, not on the front.

You can see more to this way of access in the experiences and comparison of old/new boards in this discussion:
https://github.com/Bushmills/Anycubic-Kobra-3-rooted/discussions/5

Would be wonderful if there's a way to modify the firmware to allow downgrading to any previous version via front USB flash procedure, but I am not specifically a coder, as such out of my possibilities. This would as well allow a full re-flashing of the same version to restore it to a working configuration without much hassle.

As long as the "simple" downgrade isn't possible and the version stays the same, even after flashing an older version of the OFW (you can see this in the touchscreen printer information display, as well), I'll for now not upgrade to 2.3.5.3.

For flashing the same version of the OFW over DuckPro or vice versa: That replaces several files, but doesn't do a full flush, as well doesn't necessarily replace faulty (modified) files.

AnyCubic is unfortunately in those regards very unfriendly regarding the open source / modding communities.
That's easily proven by the fact, that on newer boards the onboard USB connector SMD parts were removed to block user's access to this way.

If they were behind the open source idea like they announced often enough: "Firmware will be open in the future!" - they wouldn't do this. They released a part of the Kobra 3 firmware on their git, but not all necessary parts for full customization and ACE usage (without reverse engineering). Really a shame.

OoohPossum, 2024-11-07:
> Good news I've made it from the new Anycubic back down to the Duck :-). Tough fight but now at least the mesh and the Off-Z are right again. Maybe the topic with
> 
> * M73
> * Camera integration
> * no indication of the order on the display of the K3
> * no more communication via Anycubic to the Next Slicer
> 
> can still be fixed, that would be wonderful.

Did you complete a downgrade of the firmware or did you adapt the DuckPro 04.10.24-2 to the OFW 2.3.5.3 to result in a working combination?

If you downgraded it successfully, does your printers touchscreen device information menu show you the older version numbers? That would be a very new result,

utkabobr, 2024-11-07:
> Hi there,
> 
> Yes, I downgraded to version 2.2.9 via USB, using the menu on the K3. Then I renamed the FW from Duck to "update" and also ran it from the K3 under the machine information via touch! I think that upgrading to the new Anycubic version did not delete the data from the Duck. Because: After updating the Duck, the "old" printer.cfg I customized was still on the printer, including the old mesh default.
> 
> Greetings

DuckPro's printer.cfg is located under different path, so it remains unchanged when you go back to OFW

wacksparrow, 2024-11-07:
@utkabobr are you planning to integrate any of the good changes from 2.3.5.3 to duckpro or modifying duckpro to be able to install on 2.3.5.3? Do you have anywhere we can donate $ to compensate you for your hard work?

utkabobr, 2024-11-07:
> @utkabobr are you planning to integrate any of the good changes from 2.3.5.3 to duckpro or modifying duckpro to be able to install on 2.3.5.3? Do you have anywhere we can donate $ to compensate you for your hard work?

I'm planning on updating to 2.3.5.3, but I need to find a way of flashing mcu-s by installer first. Donations link are in readme if needed

nighty001, 2024-11-07:
@utkabobr 

Top,

I'll take care of it right away ;-) 
Thanks, you are doing a great JOB!

OoohPossum, 2024-11-07:
> Hi there,
>
> Yes, I downgraded to version 2.2.9 via USB, using the menu on the K3. Then I renamed the FW from Duck to "update" and also ran it from the K3 under the machine information via touch! I [...]

Hello @nighty001 ,

ah, I see - no full overwrite. Yes, partly overwritten re-flashing works, did that as well, but even if you re-flash the same OFW, some previously changed (OFW-)configurations will stay there, still configured.

The aspect which wonders me is that - no matter which OFW version I flashed over either DuckPro or OFW - the version didn't downgrade to an older version, but basically kept the newer version.

If you found a way to actually make the OFW printer display information menu show a lower version number after flashing a lower version OFW file, that is very interesting, would solve the no downgradeable status quo.

I tried it with different "earlier" OFW files and even if the flash was fully successful, the version wasn't lowered. That's where my curiosity comes from.

xmucio, 2024-11-25:
Hi, thank You for your work. Any progress with 2.3.5.3? I have seen sv01a's pull request, but its closed. Can You explain why? Using translator it is not understandable. Can I help somehow?

zda25, 2024-11-28:
Hello. Is it possible to flash the printer to the old firmware? So that DuckPro can be delivered

zda25, 2024-11-28:
Great news, thanks.

zda25, 2024-11-28:
Great news, thanks.

Peschi90, 2024-11-28:
Perfect news, can't wait to test ist finally 😍

zda25, 2024-11-28:
It 's hard without the usual Klipper . After switching to Klipper, will the ability to send to print from Sanicubic Slicer Next be preserved?

utkabobr, 2024-11-28:
> It 's hard without the usual Klipper . After switching to Klipper, will the ability to send to print from Sanicubic Slicer Next be preserved?

I think not. But it's useless anyway, it's just an outdated reskin of PrusaSlicer/OrcaSlicer (Next is orca)

Strongbear13, 2024-11-28:
> No need to revert to older firmware, Work-in-progress commit are already pushed, now testing on 2.3.5.3

А будет открытый текст или сразу релиз?