utkabobr/DuckPro-Kobra3 #27: ACE Pro Reverse Engineering

utkabobr, 2024-11-08:
I'll post some reverse engineered info about ACE Pro here

utkabobr, 2024-11-08:
Update log 1.3.82 => 1.3.84
[gklib-ace-update.log](https://github.com/user-attachments/files/17677950/gklib-ace-update.log)

utkabobr, 2024-11-08:
Goklipper uses OTA_... command for uploading firmware to MCU's.
ACE one is OTA_FILAMENT_HUB_START ID=0/1 UPDATE_PATH=/path/to/binary_V1.2.3.bin
It seems that it involves update path parsing, so it won't flash it unless file name is valid like ACE_test_V1.3.84_some_data.bin

utkabobr, 2024-11-08:
Fun fact: dryer temperature is NOT limited in ACE Pro's firmware, it's limited on Kobra's UI. My ACE accepted any temperature possible, but.. it may fry some casing/electronics :P
Tests required anyways

ducati1198, 2024-12-14:
Do you think we will be able to use it with other printer brands if they run klipper

utkabobr, 2024-12-26:
> Do you think we will be able to use it with other printer brands if they run klipper

Added link to my driver's repo