utkabobr/DuckPro-Kobra3 #28: Goklipper backports

utkabobr, 2024-11-08:
> [!CAUTION]
> **THIS INFO IS PROVIDED AT YOUR OWN RISK. IT INVOLVES FLASHING MCU-s**

# Backport
So, it's possible to backport almost any goklipper (gklib) to lower firmware to remain all the features needed without waiting for me to update the DuckPro.

First, download stock firmware from README and unpack it with password `U2FsdGVkX19deTfqpXHZnB5GeyQ/dtlbHjkUnwgCi+w=`
Get `firmware_v1.2.3_20001234.bin` file (name modified for example), place it under somewhere. Will be /useremain/home/ytka/firmware_v1.2.3_20001234.bin for example.

Now send:
`OTA_START UPDATE_PATH=/useremain/home/ytka/firmware_v1.2.3_20001234.bin MCU=mcu`
`OTA_START UPDATE_PATH=/useremain/home/ytka/firmware_v1.2.3_20001234.bin MCU=nozzle_mcu`
It will update mcu firmwares.

Then, replace gkapi, gklib, K3SysUi, libc_helper.so, Translate with new files from OFW/app/...
Remember to set executable flag (`chmod +x`) on gkapi, gklib, K3SysUi; Goklipper won't work otherwise.

For 2.3.5.3 it's also changed cs1237, just grab new config block from new printer.cfg:
```
[cs1237]
level_pin:nozzle_mcu:PA7
dout_pin:nozzle_mcu:PA6
sclk_pin:nozzle_mcu:PA5
register:60
sensitivity:-3000
head_block_sensitivity:-300000
scratch_sensitivity:-100000
self_check_sensitivity:-400
block_filament_sensitivity:-3000
```

# Suppressing updates
It's also possible to spoof your version to any possible through /useremain/dev/version
Just edit it to any version you want, so it won't bother you updating

ducati1198, 2024-12-09:
hi. Klipper doest connect. i can log in to mainsail but klipper not orking. i can i reflash firmware_v1.2.3_20001234.bin MCU=mcu and firmware_v1.2.3_20001234.bin MCU=nozzle_mcu if i dont gave acces to klipper to input in console. thanks

Zloyfey, 2025-02-01:
Добрый день! Подскажите пожалуйста, куда и где эту команду надо отправить?
Если в командную строку, после того как подключился по ssh, то выдает "-sh: OTA_START: not found"

Now send:
OTA_START UPDATE_PATH=/useremain/home/ytka/firmware_v1.2.3_20001234.bin MCU=mcu
OTA_START UPDATE_PATH=/useremain/home/ytka/firmware_v1.2.3_20001234.bin MCU=nozzle_mcu