utkabobr/DuckPro-Kobra3 #29: USB-Ethernet

IguanaBob, 2024-11-08:
Maybe this is a feature request, but would your firmware be able to enable USB-Ethernet support? Looking at the "official" Kobra 3 firmware repo it looks like they left a lot of the kernel drivers in including net/usb sections so in theory i could be done at least for the adapters with good linux support (I could be assuming incorrectly tho). Ethernet would be much preferred over wifi for performance/reliability, and considering they went cheap and used a wifi chip that can't do 5ghz or WPA3.

utkabobr, 2024-11-11:
I don't think that I will look into the ethernet support, it's not the scope of the project. But you can try tethering your phone to see if there is a kernel support for it, then you can buy a supported adapter by the kernel, so you won't waste your money beforehand

IguanaBob, 2024-11-11:
I know the adapters work with Android, I've used them with several Android devices.

Anyway, thanks for considering. Once I get your firmware installed and I have more access I'll poke around to see if I can discover anything myself. No need to keep this open.

Thanks.

closed 2024-11-11 by IguanaBob

IguanaBob, 2024-12-06:
Just an update in case anyone else stumbles upon this.

Now that I have DuckPro installed I did some investigation. I've tried 3 different brands of USB-ethernet adapters. On all 3 when inserting a USB-ethernet adapter into the front USB ports of the Kobra 3 with dmesg I see that it doesn't recognize the device or load any drivers. Searching the filesystem for kernel modules shows the drivers included are very limited, and none seem to work with the adapters I have tried. Unless someone can compile some of the drivers in a way that can be used inside the Kobra 3's OS I don't see a way of making this possible. But, if someone finds a way please do share.

[ 3608.675687] usb 1-1.1: new high-speed USB device number 10 using xhci-hcd
[ 3608.766075] [ac]usb_parse_endpoint:272,size=0,----
[ 3608.766243] usb 1-1.1: config 2 has an invalid descriptor of length 0, skipping remainder of the config
[ 3608.766418] [ac]usb_parse_endpoint:272,size=0,----
[ 3838.870786] usb 1-1.1: USB disconnect, device number 10
[ 3852.132316] usb 1-1.2: new high-speed USB device number 11 using xhci-hcd
[ 3852.313576] [ac]usb_parse_endpoint:272,size=0,----
[ 3852.362005] [ac]usb_parse_endpoint:272,size=0,----
[ 3852.409973] [ac]usb_parse_endpoint:272,size=0,----
[ 3858.532308] usb 1-1.1: new high-speed USB device number 12 using xhci-hcd
[ 3858.622756] [ac]usb_parse_endpoint:272,size=0,----
[ 3858.622960] [ac]usb_parse_endpoint:272,size=0,----
[ 3858.626940] [ac]usb_parse_endpoint:272,size=0,----