utkabobr/DuckPro-Kobra3 #30: Установка на 2.3.5.3

Nigh7man, 2024-11-14:
Возможна ли установка на последнюю прошивку ? Пробовал несколько раз - безрезультатно.

FeelDiference, 2024-11-15:
попытался поставить так же на последнюю, принтер чет пискнул вначале одним пиком и вот стоит ничего не меняется.

qwurete, 2024-11-15:
Можешь немного рассказать про uart в ТГ, я уже объясню человеку с такой же проблемой

FeelDiference, 2024-11-15:
тоесть из ошибки 11407 восстановление доступно только через юарт или ssh? 
причем последнее невозможно если он отрубился от вафли?

Alexdior82, 2024-11-16:
Пишите в кубик, что после последней прошивки у вас ошибка 11407, просите последний прошивку, они вышлют патч и инструкцию, после чего принтер оживет и потребует опять обновится онлайн.

Nigh7man, 2024-11-16:
> Могу скинуть патчь и инструкцию что как сделать

Скиньте пожалуйста если вас не затруднит. Боюсь окирпичить по уарту

utkabobr, 2024-11-17:
> My printer's firmware was 2.3.5.3 and Unfortunately, the screen just keeps showing error 11407. I installed RockChip_Batch_Tool program and driver for flashing the board, and I prepared the cable. But where and how can I download the img file? Help Please

You don't need img file and a cable to restore 2.3.5.3 from swu file with 11407 case. Just grab swu file from README and place it as update.swu, flash it as usual

kws1070, 2024-11-17:
> You don't need img file and a cable to restore 2.3.5.3 from swu file with 11407 case. Just grab swu file from README and place it as update.swu, flash it as usual

Thanks reply. Sure I tried first. But some how it's not working.... Maybe I did wrong.. some kind mistake. I will try again.

kws1070, 2024-11-17:
> Thanks reply. Sure I tried first. But some how it's not working.... Maybe I did wrong.. some kind mistake. I will try again.

I think I missing someing... I'm still having the same issue. If it's okay, could you help me by taking a screenshot? I've worked with many electronic devices including smartphones things, but I feel like I'm making the same basic mistake over and over again.

sv01a, 2024-11-17:
just copied update files from new firmware and replace existing one in repo. plus checked diffs between scripts, transfer missing and voila. work like a charm

Gurban-Alimirzoev, 2024-11-17:
Can you put the resulting binary somewhere or upload a commit with the changes?

Gurban-Alimirzoev, 2024-11-17:
> just copied update files from new firmware and replace existing one in repo. plus checked diffs between scripts, transfer missing and voila. work like a charm

Can you put the resulting binary somewhere or upload a commit with the changes?

Alexdior82, 2024-11-24:
[kobra_3_boot_exception_patch.zip](https://github.com/user-attachments/files/17894184/kobra_3_boot_exception_patch.zip)

Включите принтер и когда появится интерфейс с ошибкой, вставьте USB-накопитель в USB-порт принтера. Подождите около 10 секунд после двух звуковых сигналов, затем выключите принтер и извлеките USB-накопитель. Перезапустите принтер.

FeelDiference, 2024-11-24:
Это фикс для ошибки 11407?

> 

> [kobra_3_boot_exception_patch.zip](https://github.com/user-attachments/files/17894184/kobra_3_boot_exception_patch.zip)
>
> Включите принтер и когда появится интерфейс с ошибкой, вставьте [...]

Alexdior82, 2024-11-25:
Да

Isser-yts, 2024-12-12:
> Пишите в кубик, что после последней прошивки у вас ошибка 11407, просите последний прошивку, они вышлют патч и инструкцию, после чего принтер оживет и потребует опять обновится онлайн.

Что если ошибки нет, но есть сообщение после установки DuckPro поверх последней версии заводской прошики (2.3.5.3): Non official firmware code has  been detected, which damage  the printer. Please use it with  caution  
OK.
Это кирпич или ещё можно воскресить и сделать нормально? Надеюсь, живой)

utkabobr, 2024-12-12:
> Что если ошибки нет, но есть сообщение после установки DuckPro поверх последней версии заводской прошики (2.3.5.3): Non official firmware code has been detected, which damage the printer. Please use [...]

Это точно живой. Обычно решается очисткой mutable файлов или установкой свежей сборки поверх текущего состояния

Isser-yts, 2024-12-12:
> Это точно живой. Обычно решается очисткой mutable файлов или установкой свежей сборки поверх текущего состояния

То есть подождать, когда новый DuckPro выпустится? Или просто установить заводскую прошивку пока, а-то даже не попробовал печатать через неё. Обязательно новая нужна или без разницы? Как почистить mutable файлы не знаю🫠

Isser-yts, 2024-12-17:
> [kobra_3_boot_exception_patch.zip](https://github.com/user-attachments/files/17894184/kobra_3_boot_exception_patch.zip)
>
> Включите принтер и когда появится интерфейс с ошибкой, вставьте [...]

Что делает этот патч? Только лишь от ошибки 11407 спасает? А от предупреждения, что прошивка неофициальная спасёт? Стоит ли проверять?

Simfony-star, 2024-12-26:
> Обычно решается очисткой mutable файлов

А как это сделать ?

Isser-yts, 2024-12-26:
> > Обычно решается очисткой mutable файлов
> 
> А как это сделать ?

Закинь скрипт как прошивку, есть в репозитории.

Aleksandr-Sasha, 2025-01-26:
Здравствуйте. Не могу запустить веб камеру. Подскажите, что надо сделать?