utkabobr/DuckPro-Kobra3 #34: Stuck on boot logo after upgrade

Man-in-Black, 2024-11-23:
Hey,

I've tried to update my printer to your firmware, but since I tried it I'm stuck at the bootlogo.
I've waited several hours, nothing happened.

Is there a way to fix this?

Thanks.
Man-in-Black

Alexdior82, 2024-11-24:
[kobra_3_boot_exception_patch.zip](https://github.com/user-attachments/files/17894171/kobra_3_boot_exception_patch.zip)

Включите принтер и когда появится интерфейс с ошибкой ,вставьте USB-накопитель в USB-порт принтера. Подождите около 10 секунд после двух звуковых сигналов, затем выключите принтер и извлеките USB-накопитель. Перезапустите принтер.
Потом заходишь. В сведения принтера и обновляешь прошивку полностью. Отпишись как все прошло.

Man-in-Black, 2024-11-25:
Thanks for the patch.
I've tried it, but nothing happened. I've also waited abound 30 minutes, but there are no Beeps or something like that.
I've put the USB stick in, turned on the printer and waited.
I also tried to turn on the printer and then put in the USB stick.
Either way, nothing happened.

Looking for other tricks to fix it.

Alexdior82, 2024-11-25:
Не переименовывая папку закинуть на отформатированную флэшку fat 32. И как появится ошибка тогда вставить и подождать.

Man-in-Black, 2024-11-25:
I've done this exaktly like that.
But I do not have an error, just the book logo forever.
Or do I need to wait? How long can this be until an error occurs?

kws1070, 2024-11-25:
I tried this file too but it doesn't work.
A. Printer On -> Error code:11407 -> USB insert = No response
B. USB insert -> Printer On -> Error code:11407 = No response

[image]
[image]

Fat32 G:\aGVscF9zb3Nf\update.swu

Man-in-Black, 2024-11-25:
I didn't any error code :( Just the boot logo.

Man-in-Black, 2024-11-26:
@Art531 it should have been 2.3.5.3 (as stated in the slicer from the last working connection).
But I'll give it a try.

kws1070, 2024-11-26:
> [kws1070](https://github.com/kws1070), [Man-in-Black](https://github.com/Man-in-Black), what was firmware version before this failed update? Try to rename 'aGVscF9zb3Nf' folder into 'update'.

same 2.3.5.3

Man-in-Black, 2024-11-27:
I tried everything now, but it didn't work.
I'm going to return the printer within the return period and buy a new one. Luckily still the same price ;)
But I also will be happy if I can get this to work somehow if I get this error again in the future.

m3rguez0o, 2024-11-30:
Hi,
I just have the same problem as Man-in-Black...
Tried almost everything...and still stuck at the boot logo but it doesn't boot...
Before that I had the error 11407, only access on the mainsail http thing but with an error, it wasn't able to connect with moonraker so tryied to reflash with the original 2.3.5.3 and then it bricked...
I tryied to connect with an usb cable while booting on recovery with the switch sw2 maintained, the screen then don't show anything instead of the anycubic logo but i can't see nothing on the usb.
Don't really know what to do, I'd like some help if possible please before I buy a new motherboard...

kws1070, 2024-12-02:
I want buy new motherboard. But I don't know where can I buy. Someone can send link please?

Man-in-Black, 2024-12-02:
I've bought a new printer now (since they're were on sale) and got it on saturday.
After turning it on everthink worked fine, but my prints looked horrible.
So after some testing I changed the motherboard from the new printer into the old one and now everything is working again (new mainboard, old hardware).
The old printer is going to be returned, so I had no losses.

roland-pierzga-MP, 2024-12-09:
After a boot logo stack try to put firmware again to usb stick and reflash it. for me that help.

igorir3, 2024-12-28:
Damn, I have endless downloads too. As I understand it, it's useless to wait for the load to complete. The main problem is that to install the "update" it is necessary for the OS to boot and accept the file, but it does not load, therefore it cannot be updated.

In theory, you can try to "push" the distribution somehow... If at all possible

igorir3, 2025-01-09:
In short, the instructions include instructions for "updating". 

If you insert a flash drive with a normal OS before turning it on, an error will pop up during power-on.

After that, I pulled out the flash drive, copy the update file AGAIN (It was constantly erased) and put it back in

Well, it helped.
In theory, you can experiment and insert it at different points in time if my option didn't work out.