utkabobr/DuckPro-Kobra3 #35: Insufficient Local storage capacity

Maddinschluckt, 2024-11-26:
I tried to install the cfw. Pluged stick into the printer go to printer information. And tried to install it, but it says "Insufficient Local storage capacity"

So i checked and delete all flies from the storage of the printer. But still this error message

joelpr1234, 2024-12-01:
I noticed one of my 2 kobra 3 has the storage more used than the other. The difference between the 2 is that I started to flash the firmware to one, failed, reinstalled stock firmware, then I tried multiple times to install duckpro until I got it working, on the other duckpro installed fine, I practiced with the first. the one I tried multiple times to install duckpro is the one with less storage space. 

In my opinion, duckpro is adding hidden folders to the storage with the firmware files, every time you try to install the firmware a new folder is created, not sure how to fix this, maybe with ssh and deleting files manually.

weekendkoder, 2024-12-06:
I got the same issue.
What helped me: ...free space. So i needed to delete ALL *.gcode files (history files) that are stored on the internal media. It does it always, when you print via USB. After the clean-up, i was able to install the CFW.