utkabobr/DuckPro-Kobra3 #37: Error Code 11407 - solved.

weekendkoder, 2024-12-03:
**What was my status:**
- I have had installed the DuckPro FW v2.3.5.3 (DuckPro_051024_02) and everything runs fine.

**What did i done, that results into a big problem:**
- Since i upgraded to DuckPro v2.3.5.3 & everything runs smooth, i got the idea to modify the file "printer.cfg".
  Why did i wanted to modify the "printer.cfg"? My motivation was to disable the "cleaning nozzle" process that starts 
  always at the beginning of every new print. Through access by MainSail, i was able to do so. At the end-lines of 
  the "print.cfg" there were macros stored, that execute some gcode. And there were two "cleaning sections" that i 
  commented out. So i believed it should work. I saved the modified "printer.cfg" and did a reboot of the printer. And 
  this is how the catastrophe started. I got stuck while booting the printer, with the error code 11407. So i read the 
  information at the main github page of DuckPro 
  (https://github.com/utkabobr/DuckPro-Kobra3?tab=readme-ov-file#help-im-stuck-on-11407) 
  what i need to do, to solve the problem (copying stock firmware & later DuckPro FW). It did not help 
  at all, sadly.   

**How did i solve this issue:**
- Thanks to the user **KungPhoo** & his same experience & information's i was able to fix it. You need to flash the 
  printer with a package-patch by using a USB. This patch comes from the official Anycubic site. Funny thing is, that 
  this patch is designed to solve the "error code 10408". But somehow it works also with the "error code 11407". 
  Here is the link to the patch: https://wiki.anycubic.com/en/error-codes/10408-code/k3 
  
I hope my told journey can help other kobra 3 users, which use the current DuckPro v2.3.5.3 FW and have this described problem. 
So be carefull when editing the file "printer.cfg". I dont know why commenting out something, result in such a big problem.

kws1070, 2024-12-03:
Unfortunately, this did not solve my problem. Anyway, thanks for sharing the information. :)

Peschi90, 2024-12-03:
You can also connect with filezilla to the file structure of the printer and change back the printer.cfg to a working status. Then you don't need to flash anything new.

kws1070, 2024-12-03:
> You can also connect with filezilla to the file structure of the printer and change back the printer.cfg to a working status. Then you don't need to flash anything new.

If I use FileZilla, what is the connection port number? If possible, can you record the connection process and show me? The slicer program says the printer is working (of course, the printer does not respond at all), but I can't even find the IP address of the printer.

Peschi90, 2024-12-03:
You have to find the IP address in your router.

kws1070, 2024-12-03:
> You have to find the IP address in your router.

Sure. I know. I checked router. Just can't find address..

weekendkoder, 2024-12-03:
@Peschi90 . Thank you for the hint. I remember that i tried to connect to my printer while having this error, via "MainSail". MainSail still worked, but said: no connection to Klipper (so i felt lost). Hm, i guess this means the ip-adress works & is active. But i did not came up with the idea to try out filezilla. Next time i will try to reproduce the error. Once done, i will use the filezilla solution.

weekendkoder, 2024-12-03:
**Update:**

I was able to use the Stock Firmware. But if i try to flash the DuckPro CFW (DuckPro_051024_02.zip or DuckPro_031224.zip) i will stay at the error "code 11407". Seems like there is no way, to install this CFW. My guess ist, that something dramaticaly changed the boot-sequence, at least for DuckPro FW. Still dont understand why a printer.cfg modification was able to make this happen. 

**in addition:**
Even i went back to Stock FW, it does not work properly. No communication. Doesn´t matter which action you want to start. Print, calibration, etc. Nothing works. But the Stock Firmware is still available, via LCD. 

**Error Report from MainSail:**
_Moonraker can't connect to Klipper!
Please check if the Klipper service is running and klippy_uds_address is correctly configured in the moonraker.conf._

I think it has nothing to do with "moonraker.conf". I have connection to the printer, via MainSail. I hope i have a chance to modify some files deep inside the system to solve this problem. If someone has an idea, just comment.

utkabobr, 2024-12-03:
> **Update:** I was able to use the Stock Firmware. But if i try to flash the DuckPro CFW (DuckPro_051024_02.zip or DuckPro_031224.zip) i will stay at the error "code 11407". Seems like there is no [...]

Edit /userdata/app/gk/config/api.cfg. Change log level from "error" to "debug". Check logs in /tmp/gklib.log

weekendkoder, 2024-12-03:
@utkabobr 

> Edit /userdata/app/gk/config/api.cfg. Change log level from "error" to "debug". Check logs in /tmp/gklib.log

Hm, how to access these files ? Or should i edit the files localy and re-pack the "update.swu" & than flash again ?

utkabobr, 2024-12-03:
Via SSH/ADB

weekendkoder, 2024-12-03:
Is this the correct command line for ssh, to connect to the printer ? :

ssh root@xxx.xxx.xxx.xxx  (ssh command with your IP)

user: root
pass: rockchip

And like at the main-github-page described, above the user & password.

utkabobr, 2024-12-03:
Yup

weekendkoder, 2024-12-03:
Ok. I followed the steps. Editing "api.cfg" and reading "gklib.log".

After making changes to the "api.cfg" i did a reboot of kobra 3. But nothing happend. It stays the same, with the same error: "error code 11407". Here is the "gklib.log" report:

[image]

utkabobr, 2024-12-03:
You should edit cs1237 block, check out info about backports in pinned issue

weekendkoder, 2024-12-03:
Okay. Sorry i am not good with linux. I guess you mean to edit the file "cs1237.go" ? (inside: /data/liuxiaobo/k3_sys28/1.1/k3c/project/
[cs1237.go]). I checked the path, but it dont exists. Strange. And than i should edit it ... the block is 57 & 351, i guess ? Inside this file (these blocks) i should search for the "backports" that are the issue? I dont know what is meant by "backports".

Maybe you can explain it a bit more in detail to me. And maybe you can check if there is the folder "liuxiaobo" (/data/liuxiaobo). What i cannot find. Thank you.

utkabobr, 2024-12-03:
No. Printer.cfg [cs1237]

weekendkoder, 2024-12-03:
K...got it. I found the [cs1237] Block inside the "printer.cfg" file. This is the one provided with DuckPro FW:

[cs1237]
#level_pin:nozzle_mcu:PA7
dout_pin:nozzle_mcu:PA6
sclk_pin:nozzle_mcu:PA5
register:60
sensitivity:-3000
samples_count:50
data_bit:24
head_block_sensitivity:-300000
scratch_sensitivity:-100000
self_check_sensitivity:-400
block_filament_sensitivity:-3000

The one, i had inside my "printer.cfg", which causes problems looks like this:
[cs1237]
level_pin:nozzle_mcu:PA7
dout_pin:nozzle_mcu:PA6
sclk_pin:nozzle_mcu:PA5
register:60
sensitivity:-3000
head_block_sensitivity:-300000
scratch_sensitivity:-100000
self_check_sensitivity:-400
block_filament_sensitivity:-3000

As you can see, some lines are missing. I don´t know exactly why. I did not removed them. My only guess is, that this happened while i was flashing the original stock FW with DuckPro FW and vise versa. But i cant say for sure how it happened. Seems like this causes the "error code 11407". For now the printer works again with our custom DuckPro FW. I will doe some more tests, and will report later. 

Big Thanks to @utkabobr for all the support.

utkabobr, 2024-12-03:
New printer.cfg is provided with the update for 2.3.5.3, and it should migrate to new one as stated in release notes

weekendkoder, 2024-12-04:
Okay...i wanted to proof if it works again. Like:

1. go back to original stock FW v2.3.5.3
   (it does appear, but nothing work. No ACE connection, nor PrintHeat Connection)
2. upgrade to DuckPro "GitHub_Release_03.12.24_(DuckPro_031224)"
   Seems to upgrade, but i stuck at the boot-logo (loading bar, don´t disappear anymore)
3. tried to go back to "GitHub_Release_05.10.24-02_(DuckPro_051024_02)"
   It does not accept this "update.swu". After 10min i disconnet the usb and i culd see
   this file was not deleted by the system. After reboot the same situation: stuck at the 
   boot-logo screen
   
Hm, now i dont have any clue what to do. I can try to read some logs again. Don´t know which are relevant.

xmucio, 2024-12-04:
I tried to change bed mesh from 5,5 to 8,8 and the same happend - stuck at boot logo with full bar. I fix it by reflashing CFW. It seems that any change in printer.cfg is causing a crash.

weekendkoder, 2024-12-04:
@xmucio , thank you for sharing your experience. May i ask, which CFW you re-flashed ? (i tried both DuckPro´s: 05.10.24-02_(DuckPro_051024_02) & 03.12.24_(DuckPro_031224). But no luck. Even Stock FW did not worked)

> It seems that any change in printer.cfg is causing a crash.

For me, changing or better say "commenting out" stuff, like the cleaning procedure worked. Would like to test more, but i stuck...

Anguysh, 2024-12-05:
Can you access mainsail? Some changes to printer.cfg can cause an error while others work. I just go back into mainsail and change back what I changed in printer.cfg.

xmucio, 2024-12-05:
After changing bed mesh i was unable to change it back via mainsail. There was no connection with Klipper and after saving and restarting the file remain the same values.

@weekendkoder , i used the last release.

utkabobr, 2024-12-05:
printer.cfg can be changed via SSH/ADB if it doesn't boot on screen. You must've changed points on lagrange, it's not supported in Klipper. You should change to bicubic

xmucio, 2024-12-05:
What is the path of printer.cfg visible in mainsail? I found one via ssh in root@Rockchip:/userdata/app/gk#, but it has stock values. It has to be somewhere else, but i couldn't find it.

utkabobr, 2024-12-05:
/useremain/home/ytka/printer_data/config

utkabobr, 2024-12-05:
> I am not able to enter/usermain via ssh.
> 
> `root@Rockchip:/userdata/app/gk# cd /usermain -sh: cd: can't cd to /usermain`

You missed "e" :)

xmucio, 2024-12-05:
Lol, haha... Tried at least 20 times without luck. Thank You, such a stupid mistake. 

I will try to use bicubic algorithm. I have decent bed, but still first layer sucks. I had a lot of better results on my previous Kobra 2 neo with Marlin. I think there is a issue with auto levelling. 

[image]

ducati1198, 2024-12-13:
This is the picture when ingot the error

[image]

weekendkoder, 2024-12-14:
@ducati1198 ...wow, that´s a good hint. Very interesting. Sound also logical, if he printhead-mcu does not have the latest FW, the Main-MCU with the latest FW will have trouble to communicate with the print-head. I will try it out these day, and report back. Thank you.

ducati1198, 2024-12-14:
> @ducati1198 ...wow, that´s a good hint. Very interesting. Sound also logical, if he printhead-mcu does not have the latest FW, the Main-MCU with the latest FW will have trouble to communicate with the print-head. I will try it out these day, and report back. Thank you.

You can even go back to 1.1.6 or 1.24 and vice versa