utkabobr/DuckPro-Kobra3 #39: Stuck on boot logo

supersailor89, 2024-12-04:
Stuk on boot logo after update. i can see it on the slicer app.  and i connect to wifi but it does not do anything

weekendkoder, 2024-12-04:
I have the same situation now. I am not sure, which steps should be made. I tried f.e. to go back to Stock FW. It seems to work, but it has not functionality, like no ACE connection, no PrintHead connection. Did you tried to go back to stock FW ?

supersailor89, 2024-12-04:
https://wiki.anycubic.com/en/error-codes/10408-code/k3

I fixed following this

weekendkoder, 2024-12-04:
Thx....i tried this allready. No luck this time. Somehow after the new Version of DuckPro i stuck at boot.

utkabobr, 2024-12-05:
> Stuk on boot logo after update. i can see it on the slicer app. and i connect to wifi but it does not do anything

Seems like printer.cfg is wrong. Check it via Mainsail/SSH/ADB, especially [cs1237] block

supersailor89, 2024-12-05:
> Thx....i tried this allready. No luck this time. Somehow after the new Version of DuckPro i stuck at boot.

Mine was stuck in boot logo, then I tried to revert back to stock firmware. To the 2.3.3, after that I got the error code. Then I did the patch. After that it booted and updated

supersailor89, 2024-12-05:
> > Stuk on boot logo after update. i can see it on the slicer app. and i connect to wifi but it does not do anything
> 
> Seems like printer.cfg is wrong. Check it via Mainsail/SSH/ADB, especially [cs1237] block

I shows mainsail, but can't get in the web with or ssh. But it's fixed now

weekendkoder, 2024-12-05:
@supersailor89 ...hm, if you say 2.3.3, it could mean more than one vers. We have these FW´s:
- v2.3.3.2
- v2.3.3.9
- v2.3.5.3

My guess is, you flashed the "v2.3.3.9" FW ?

weekendkoder, 2024-12-05:
I recovered my Printer from Boot-Stuck issue. Thanks to @supersailor89 for the hint.
I used the **Stock FW version 2.3.3.9**. Stock v2.3.5.3 did NOT worked for me.

Now i am back at DuckPro v2.3.3.9. If i update to DuckPro / Stock v2.3.5.3 hte problems would start again.
Therefore i stay with the older custom-FW ... for now.

supersailor89, 2024-12-05:
> I recovered my Printer from Boot-Stuck issue. Thanks to @supersailor89 for the hint. I used the **Stock FW version 2.3.3.9**. Stock v2.3.5.3 did NOT worked for me.
>
> Now i am back at DuckPro [...]

You welcome, glad you are up and running.

Steven261092, 2024-12-05:
"Hey everyone, I’m experiencing the same issue. When I install the latest release of DuckPro with the current firmware version 2.3.5.3, I get stuck on the boot screen. I can access Mailsail, but there’s no connection to Klipper.

I’ve read through some of the posts here and tried downgrading to 2.3.3.9, but that didn’t work at all.

Does anyone have an idea?"

weekendkoder, 2024-12-05:
Hm ... Maybe you can try first to go back to STOCK v2.3.5.3 and than back to v2.3.3.9. Something else you should consider:
- by booting up with USB it takes a lot of time, until the printer reboot and flash its FW internaly.

So this means it took me 2x reboots (automatic reboots by kobra3) and only after the 2nd reboot the new FW is flashed. Maybe you are not waiting long enough ? F.e. if you turn off the printer too fast, it would disturb the flashing / copying process i guess. So try to flash again, and give it a bit more time.

One small hint is, that when nothing seems to happen, after round about 1min you will hear a chip-sound from the old Game DOOM (DOS version). And after this chipssound, the kobra3 will reboot again automaticaly. I did not count the seconds / minutes, but just try to be more patient. It could help ... i hope.

Steven261092, 2024-12-05:
"The sound comes immediately after I turn on the printer... Then it restarts once. At some point, a message appears on the display saying that it is unofficial firmware. After that, nothing else happens, even if I let the printer run like that for an hour. There’s no second restart or anything else.

And yes, I tried to downgrade from stock to stock, but unfortunately, that doesn’t work either

utkabobr, 2024-12-05:
> "The sound comes immediately after I turn on the printer... Then it restarts once. At some point, a message appears on the display saying that it is unofficial firmware. After that, nothing else [...]

Try that, DO NOT downgrade: https://github.com/utkabobr/DuckPro-Kobra3/releases/tag/cleanup_mutable_config

weekendkoder, 2024-12-05:
What is this file "printer_mutable.cfg" doing ?

utkabobr, 2024-12-05:
> What is this file "printer_mutable.cfg" doing ?

It's used to store input shaping and bed mesh parameters. You could just recalibrate after that

Steven261092, 2024-12-06:
@weekendkoder  so now it works ... it was just the other printer cfg ...

supersailor89, 2024-12-06:
New duck pro works now

IguanaBob, 2024-12-06:
I've run into this too.

This Kobra 3 is brand new. I haven't even tried printing or doing anying special with it yet, so it should be pretty close to stock. All i've done so far:

Initial setup on stock firmware 2.3.3.9, including bed leveling, vibration check, etc.
Connect to wifi. Pair with Anycubic app on my phone.
Updated firmware on ACE to newest.
Updated firmware on Kobra 3 to 2.3.5.3.
Installed DuckPro unzipped DuckPro_051224.zip using the red USB that came with the printer reformatted fresh with FAT32 from a Linux computer.
It beeped, played the Doom song, rebooted a couple times, showed the "Non Official" warning message on screen, but then got stuck at the boot logo. After waiting a long time I power cycled, but it got stuck on boot screen again.
I then tried installing from CleanupMutableConfig.zip. At the boot screen if I plug in the USB it beeps, the light on the USB flashes a few times, but then nothing happens for a long time. Eventually I try restarting the printer but it still gets stuck at the boot screen. If I re-insert the USB it beeps again but still the same result.

Should I try re-flashing to stock then starting over?

utkabobr, 2024-12-06:
> I've run into this too.
>
> This Kobra 3 is brand new. I haven't even tried printing or doing anying special with it yet, so it should be pretty close to stock. All i've done so far:
>
> Initial setup on [...]

 Just flash this, should help: https://github.com/utkabobr/DuckPro-Kobra3/releases/tag/cleanup_mutable_config

utkabobr, 2024-12-06:
> That was what I tried
> 
> > I then tried installing from CleanupMutableConfig.zip.
> 
> Sorry for confusion.

Ah, I see. So mainsail shows up or not?

IguanaBob, 2024-12-06:
> > That was what I tried
> > > I then tried installing from CleanupMutableConfig.zip.
> > 
> > 
> > Sorry for confusion.
> 
> Ah, I see. So mainsail shows up or not?

Mainsail yes, but it can't connect to moonraker

utkabobr, 2024-12-06:
> > > That was what I tried
> > > > I then tried installing from CleanupMutableConfig.zip.
> > > 
> > > 
> > > Sorry for confusion.
> > 
> > 
> > Ah, I see. So mainsail shows up or not?
> 
> Mainsail yes, but it can't connect to moonraker

Moonraker or Klipper? It's important as if it can connect to moonraker, then we can change config files

IguanaBob, 2024-12-06:
> > > > That was what I tried
> > > > > I then tried installing from CleanupMutableConfig.zip.
> > > > 
> > > > 
> > > > Sorry for confusion.
> > > 
> > > 
> > > Ah, I see. So mainsail shows up or not?
> > 
> > 
> > Mainsail yes, but it can't connect to moonraker
> 
> Moonraker or Klipper? It's important as if it can connect to moonraker, then we can change config files

Looks like Moonraker to me?
[image]
[image]

IguanaBob, 2024-12-06:
If I hit Try Again a few times I finally get this message.

> If this message appears repeatedly, please have a look in the log file located at:
>
>    ~/printer_data/logs/moonraker.log

I'll see if I can get to that file through ADB when I have time.

IguanaBob, 2024-12-06:
Bottom of /useremain/home/ytka/printer_data/logs/moonraker.log

```-------------------- Log Start | Thu Jan  1 00:00:23 1970 --------------------
platform: Linux-5.10.160-armv7l-with-libc0
data_path: /useremain/home/ytka/printer_data
is_default_data_path: True
config_file: /useremain/home/ytka/printer_data/config/moonraker.conf
startup_warnings: []
verbose: False
debug: False
asyncio_debug: False
is_backup_config: False
is_python_package: False
instance_uuid: 71e8dfd893b04983ae99a94001f40b17
unix_socket_path: /useremain/home/ytka/printer_data/comms/moonraker.sock
software_version: ?
log_file: /useremain/home/ytka/printer_data/logs/moonraker.log
python_version: 3.11.10 (main, Sep 12 2024, 17:32:54) [GCC 8.3.0]
launch_args: /useremain/dist/bin/python3 /useremain/home/ytka/moonraker/moonraker/moonraker.py
msgspec_enabled: False
uvloop_enabled: False
1970-01-01 08:00:23,548 [server.py:main()] - Server Config Error
Traceback (most recent call last):
  File "/useremain/home/ytka/moonraker/moonraker/confighelper.py", line 973, in _parse_file
    stat = file_path.stat()
           ^^^^^^^^^^^^^^^^
  File "/useremain/dist/lib/python3.11/pathlib.py", line 1013, in stat
    return os.stat(self, follow_symlinks=follow_symlinks)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/useremain/home/ytka/printer_data/config/moonraker.conf'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/useremain/home/ytka/moonraker/moonraker/server.py", line 661, in main
    server = Server(app_args, log_manager, event_loop)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/useremain/home/ytka/moonraker/moonraker/server.py", line 86, in __init__
    self.config = config = self._parse_config()
                           ^^^^^^^^^^^^^^^^^^^^
  File "/useremain/home/ytka/moonraker/moonraker/server.py", line 163, in _parse_config
    config = confighelper.get_configuration(self, self.app_args)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/useremain/home/ytka/moonraker/moonraker/confighelper.py", line 1088, in get_configuration
    source.read_file(start_path)
  File "/useremain/home/ytka/moonraker/moonraker/confighelper.py", line 1076, in read_file
    self._parse_file(main_conf, [])
  File "/useremain/home/ytka/moonraker/moonraker/confighelper.py", line 1061, in _parse_file
    raise ConfigError(
moonraker.confighelper.ConfigError: Configuration File Not Found: '/useremain/home/ytka/printer_data/config/moonraker.conf''
1970-01-01 08:00:23,554 [server.py:main()] - Server Shutdown
```

```
root@Rockchip:/useremain/home/ytka/printer_data/logs# ls -l /useremain/home/ytka
/printer_data/config/moonraker.conf
ls: /useremain/home/ytka/printer_data/config/moonraker.conf: No such file or directory
```

```
root@Rockchip:/useremain/home/ytka/printer_data/logs# find / -iname "moonraker.c
onf"
/useremain/home/ytka/moonraker/docs/moonraker.conf
```

utkabobr, 2024-12-06:
Upload printer_data from repo via adb

IguanaBob, 2024-12-06:
Yup, config is missing there. I'll try to adb push that.

```
root@Rockchip:/oem# find /useremain/home/ytka/printer_data/
/useremain/home/ytka/printer_data/
/useremain/home/ytka/printer_data/misc
/useremain/home/ytka/printer_data/comms
/useremain/home/ytka/printer_data/.moonraker.uuid
/useremain/home/ytka/printer_data/logs
/useremain/home/ytka/printer_data/logs/moonraker.log
```

IguanaBob, 2024-12-06:
A quick git clone and adb push

```$ git clone https://github.com/utkabobr/DuckPro-Kobra3.git
$ adb push DuckPro-Kobra3/files/home/ytka/printer_data/ /useremain/home/ytka/
DuckPro-Kobra3/files/home/ytka/printer_data/: 7 files pushed. 0.3 MB/s (737594 bytes in 2.507s)
```

And success! It boots and printer info shows "DuckPro // Anycubic Kobra 3" for Machine

Thanks for the help.

IguanaBob, 2024-12-06:
I happened to notice SSH still wasn't running, and poked around more via ADB and see that other stuff in /useremain/ like openssh and files in /app/ don't match what I see on git. I pushed files/app and files/openssh as well but ssh doesn't want to start, so something else must be missing. Maybe some data is missing from  DuckPro_051224.zip?

utkabobr, 2024-12-06:
Just reboot, start.sh copies everything into the place

IguanaBob, 2024-12-06:
> Just reboot, start.sh copies everything into the place

Ah, lol, and I was already in the process of adb push to /ac_lib/lib/openssh/ and chmod on key files. Let me wipe that out and try reboot like you say.

IguanaBob, 2024-12-06:
Just had to push to the correct location before reboot.

$ adb push DuckPro-Kobra3/files/openssh/ /useremain/home/ytka/

But, perms are not correct.

```
root@Rockchip:/# /ac_lib/lib/openssh/sshd_start.sh
/bin/sh: /ac_lib/lib/openssh/sshd_start.sh: Permission denied

root@Rockchip:/# chmod +x /ac_lib/lib/openssh/bin/*
root@Rockchip:/# chmod +x /ac_lib/lib/openssh/sbin/*
root@Rockchip:/# chmod +x /ac_lib/lib/openssh/sshd_start.sh

root@Rockchip:/# /ac_lib/lib/openssh/sshd_start.sh
config
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@         WARNING: UNPROTECTED PRIVATE KEY FILE!          @
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
Permissions 0644 for '/ac_lib/lib/openssh/etc/ssh_host_rsa_key' are too open.
It is required that your private key files are NOT accessible by others.
This private key will be ignored.
Unable to load host key "/ac_lib/lib/openssh/etc/ssh_host_rsa_key": bad permissions
Unable to load host key: /ac_lib/lib/openssh/etc/ssh_host_rsa_key
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@         WARNING: UNPROTECTED PRIVATE KEY FILE!          @
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
Permissions 0644 for '/ac_lib/lib/openssh/etc/ssh_host_ecdsa_key' are too open.
It is required that your private key files are NOT accessible by others.
This private key will be ignored.
Unable to load host key "/ac_lib/lib/openssh/etc/ssh_host_ecdsa_key": bad permissions
Unable to load host key: /ac_lib/lib/openssh/etc/ssh_host_ecdsa_key
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@         WARNING: UNPROTECTED PRIVATE KEY FILE!          @
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
Permissions 0644 for '/ac_lib/lib/openssh/etc/ssh_host_ed25519_key' are too open.
It is required that your private key files are NOT accessible by others.
This private key will be ignored.
Unable to load host key "/ac_lib/lib/openssh/etc/ssh_host_ed25519_key": bad permissions
Unable to load host key: /ac_lib/lib/openssh/etc/ssh_host_ed25519_key
sshd: no hostkeys available -- exiting.

root@Rockchip:/# chmod 600 /ac_lib/lib/openssh/etc/*_key

root@Rockchip:/# /ac_lib/lib/openssh/sshd_start.sh
config

root@Rockchip:/# ps | grep ssh
 1960 root      3960 S    sshd: /ac_lib/lib/openssh/sbin/sshd -f /ac_lib/lib/o
 2011 root      1208 S    grep ssh

root@Rockchip:/# netstat -ltn | grep 22
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN    
```

Still running after reboot, too.

IguanaBob, 2024-12-06:
In regards to my initial thought:

> Maybe some data is missing from DuckPro_051224.zip?

Unzipping the files I do see openssh under DuckPro_051224.zip/aGVscF9zb3Nf/update.swu/setup.tar.gz/openssh and the cleanup should not have removed them, so maybe my initial flash was incomplete.

supersailor89, 2024-12-06:
@IguanaBob 

I see that you said the printer was new. Did you connect to the network and update to latest firmware like the instructions said? And then flash the latest DuckPro?.

Maybe try reinstalling stock latest and the do latest DuckPro..

IguanaBob, 2024-12-06:
> @IguanaBob
> 
> I see that you said the printer was new. Did you connect to the network and update to latest firmware like the instructions said? And then flash the latest DuckPro?.
> 
> Maybe try reinstalling stock latest and the do latest DuckPro..

Yes sir. All I had done was
- Setup, level, shake test
- Connect to wifi/app
- Update ACE firmware
- Update Kobra firmware to 2.3.5.3
- Installed DuckPro

91eb0r9, 2024-12-14:
Hello. Stuck on boot logo. In mainsail i see "Moonraker can't connect to Klipper!" error. "Cleanup mutable config" script didn't worked via USB. What else can I do?