utkabobr/DuckPro-Kobra3 #45: Printer.cfg - customize specific MACRO´s

weekendkoder, 2024-12-05:
I made my prints faster (at the beginning of an print) by excluding some "preconfigured macro´s" inside the file "printer.cfg". These macro´s that i deactivated (by commenting them out) are:
- [gcode_macro SINGLE_WIPE_NOZZLE]
- [gcode_macro WIPE_NOZZLE]
- [gcode_macro UNDERLINE]

But there are much more macro´s that are under the hood, that are not included inside the file "printer.cfg". There is an Block called **[printer_marco]** (i guess it is a typewriting mistake, it should be named "macro". But it seems it is now like this inside anycubics FW) that hold´s much more preconfigurated gcodes. I am not 100% sure if it is this block, but when i comment it, to disable it, the printer dont execute any premade gcode orders. But it also dont print, because the temp decrease to 0. So its not possible just to deactivate this block.

My question is, if someone know´s how to get access to these "under the hood macro´s". I think it would be a benefit, if we figure out, how to modify them, to make the kobra 3 work like we want it to work.

joelpr1234, 2024-12-05:
You commented those 3 macros and the printer still works? I tried to change the 5x5 grid and got an error, I had to restore the 5x5 setting in order to make the printer work again. I thought if you change anything on the printer.cfg, the printer will die.

Now, what UNDERLINE does?

utkabobr, 2024-12-05:
You can change to more than 5x5, but you should change lagrange to bicubic

weekendkoder, 2024-12-05:
Every time i start a new print, i wait round about 10 minutes before the job starts (maybe shorter, but it feels like this). This is because of all these macro´s. 

1.) Heat up to 170. (this takes time)
2.) cleaning (this takes time)
3.) cool down to 140 (this takes time)
4.) making probes (this takes time)
5.) heat up to 220 (this takes time)
6.) Doing a cleaning line (this takes time)

So from my understanding _"UNDERLINE"_ is the **cleaning line** (point 6), before the desired print starts. Because i commented [gcode_macro UNDERLINE] out, it goes directly to the print job. In my opinion all this steps within the macro´s being done, it takes a ridiculous amount of time. Imagine you can skip all the steps, and just print your object. Bed mesh is stored already and Z-offset too and your nozzle is clean too. Only once z-axis needed to be probed.

joelpr1234, 2024-12-05:
> Every time i start a new print, i wait round about 10 minutes before the job starts (maybe shorter, but it feels like this). This is because of all these macro´s.
> 
> 1.) Heat up to 170. (this takes time) 2.) cleaning (this takes time) 3.) cool down to 140 (this takes time) 4.) making probes (this takes time) 5.) heat up to 220 (this takes time) 6.) Doing a cleaning line (this takes time)
> 
> So from my understanding _"UNDERLINE"_ is the **cleaning line** (point 6), before the desired print starts. Because i commented [gcode_macro UNDERLINE] out, it goes directly to the print job. In my opinion all this steps within the macro´s being done, it takes a ridiculous amount of time. Imagine you can skip all the steps, and just print your object. Bed mesh is stored already and Z-offset too and your nozzle is clean too. Only once z-axis needed to be probed. Therefore i think it would be a good idea, that the user decide which steps he needs.

Thanks for the explanation, I'm not good with code. Could you let me know how I can comment those? For example, I suppose I have to comment [gcode_macro UNDERLINE] and the entire data after. Is that correct? Thanks again.

Sol1tas, 2024-12-06:
Oh, btw, can you actually link the whole .cfg file, so I can see at least what macros're on the list? I've rebuilt my cobra on a fresh klipper install and now having trouble to configure parking and nozzle wiping macros

weekendkoder, 2024-12-06:
I am not switched to the newest DuckPro FW. So my _printer.cfg_ is from **DuckPro v2.3.3.9**. Be aware of it.

@joelpr1234  No Problem. Here you can see how i done it:
[image]
The "commented out" lines are using this "#" symbol.

@Sol1tas I attached _my version_ of "**printer.cfg**" as an *.zip attachment (printer_.mod.zip). File: [printer_(mod).zip](https://github.com/user-attachments/files/18037014/printer_.mod.zip)

tommmmmd, 2024-12-29:
hab versucht ein selbst erstelltes Macro.cfg aufzurufen aber es funktioniert nicht. was mache ich falsch?

NatroNx, 2024-12-31:
> Every time i start a new print, i wait round about 10 minutes before the job starts (maybe shorter, but it feels like this). This is because of all these macro´s.
>
> 1.) Heat up to 170. (this takes [...]

you are mostly right but the slowest part is purging and pooping two times... - also wasting matieral, but we have to deal with this for now.

What you can do is:

1.) Heat up to 170. (this takes time)    - can't make that faster
2.) cleaning (this takes time)    - while this is true the nozzle needs to be clean - but I don't understand why they let it rest in the middle of the silicone wiper. It feels like a bug. I changed it to park 8mm above the wiper
3.) cool down to 140 (this takes time) - true - I changed that to stay at 170 degree - theretically - you nozzle is more likley to ooze material while probing - but I never had issues with it and it is way faster not to wait for it to cool down and heat up again
4.) making probes (this takes time) - true
5.) heat up to 220 (this takes time) - less time if you are alreary at 170
6.) Doing a cleaning line (this takes time)  - the cleaning line almost takes no time - the purge after the line takes time

So what I changed is basically 2 lines. The temperature from 140 to 170 (just search for 140 in the printer.cfg - there is only one spot (something link end_temperature)
And at the end of the wipe script change  (in gcode_macro WIPE_NOZZLE) 
G1 X46 Y259.5 Z-0.8 F9600 to 
G1 X46 Y259.5 Z8 F9600 
this will just park the nozzle to a better position after wiping

joelpr1234, 2025-01-04:
> > Every time i start a new print, i wait round about 10 minutes before the job starts (maybe shorter, but it feels like this). This is because of all these macro´s.
> > 1.) Heat up to 170. (this takes time) 2.) cleaning (this takes time) 3.) cool down to 140 (this takes time) 4.) making probes (this takes time) 5.) heat up to 220 (this takes time) 6.) Doing a cleaning line (this takes time)
> > So from my understanding _"UNDERLINE"_ is the **cleaning line** (point 6), before the desired print starts. Because i commented [gcode_macro UNDERLINE] out, it goes directly to the print job. In my opinion all this steps within the macro´s being done, it takes a ridiculous amount of time. Imagine you can skip all the steps, and just print your object. Bed mesh is stored already and Z-offset too and your nozzle is clean too. Only once z-axis needed to be probed. Therefore i think it would be a good idea, that the user decide which steps he needs.
> 
> you are mostly right but the slowest part is purging and pooping two times... - also wasting matieral, but we have to deal with this for now.
> 
> What you can do is:
> 
> 1.) Heat up to 170. (this takes time) - can't make that faster 2.) cleaning (this takes time) - while this is true the nozzle needs to be clean - but I don't understand why they let it rest in the middle of the silicone wiper. It feels like a bug. I changed it to park 8mm above the wiper 3.) cool down to 140 (this takes time) - true - I changed that to stay at 170 degree - theretically - you nozzle is more likley to ooze material while probing - but I never had issues with it and it is way faster not to wait for it to cool down and heat up again 4.) making probes (this takes time) - true 5.) heat up to 220 (this takes time) - less time if you are alreary at 170 6.) Doing a cleaning line (this takes time) - the cleaning line almost takes no time - the purge after the line takes time
> 
> So what I changed is basically 2 lines. The temperature from 140 to 170 (just search for 140 in the printer.cfg - there is only one spot (something link end_temperature) And at the end of the wipe script change (in gcode_macro WIPE_NOZZLE) G1 X46 Y259.5 Z-0.8 F9600 to G1 X46 Y259.5 Z8 F9600 this will just park the nozzle to a better position after wiping
> 
> Forgot to mention - 4 Minutes from cool to printing with this changes - still feels long

the point 2) is not a bug, is intentional to keep the nozzle tip clean when reducing temp to avoid ozzing when probing, if some material gets oozed to the bed, this could cause a mark in the first layer, specially if you had a different color in a previous print.

there are aftermarket nozzles, smaller and they heat faster, I tried them and definitive heats faster and cools down faster, to the point that you have to pre heat the bed before doing the print because the bed temp can't reach the temp (mine 60 C for PLA) before the nozzle start probing and giving erroneous proving, the hotter, the more the bed expands.  I also noticed the temp sensors on those nozzles are off by up to +20 degrees C, horrible printing until I found they were measuring wrong with a temp tower print and a Fluke temp probe sensor, I ended installing the factory ones, they work fine. I have 2 K3 and 3 ACE.