utkabobr/DuckPro-Kobra3 #48: LAN-only

IguanaBob, 2024-12-06:
This isn't an issue, rather more of an information share.

I wanted to confirm that with DuckPro I could print without the use of Internet or Anycubic's cloud systems or app. Here's what I found.

Test: After installing DuckPro and making sure i could connect I enabled LAN-mode on the touchscreen. I also blocked access to the internet for the Kobra 3's IP address in my router. I confirmed via SSH/ADB that the printer could no loner reach the internet or was even attempting to connect to the cloud.

Result: I can still connect to Mainsail over my LAN and start print jobs. Orca can still send gcode files directly. The Anycubic app is no longer able to connect, which I wanted. The App no longer connects, so all control has to be done via Mainsail/klipper or with the touch screen, which I am totally okay with. Firmware updates are not possible without the app, though, so you would need to temporarily disable any blocks to do a firmware update should any more be release.

So, success, my Kobra 3 is now fully offline and I don't need the stupid Anycubic slicer or App. Bye bye, Anycubic Cloud!!!

utkabobr, 2024-12-06:
You can use mainsail/moonraker without disabling any of anycubic's services, they do not conflict

IguanaBob, 2024-12-06:
Sure, but I WANT to disable their services. Just sharing that it does work.

utkabobr, 2024-12-06:
Just disable gkapi in start.sh. But it's also disables touchscreen as it depends on it

IguanaBob, 2024-12-06:
Interesting, didn't know that was possible that way. I think I'd rather have the touchscreen still, I just wanted to block anything going out to the internet.

closed 2025-03-01 by IguanaBob