utkabobr/DuckPro-Kobra3 #49: Camera tutorial?

joelpr1234, 2024-12-06:
Can someone please do a dummies tutorial of how to configure the camera? Thanks

Anguysh, 2024-12-07:
I'd like this as well, tried everything.

utkabobr, 2024-12-07:
Place new file .camera in config directory (where printer.cfg), reboot the printer

ClaireRibon, 2024-12-07:
What's the addresses needed when adding a camera to mainsail? I tried default as well as others.

joelpr1234, 2024-12-07:
> Place new file .camera in config directory (where printer.cfg), reboot the printer

Thanks, I did in my two k3, now how I configure the camera?

joelpr1234, 2024-12-07:
@jbatonnet @utkabobr 
I tried this: https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/output_http/README.md
Nothing worked. I can see I have the .camera file in the config folder
Please help

tymtymm, 2024-12-07:
@joelpr1234 i don't think it's mentioned anywhere, but if you go to nginx conf file you can see that camera is at /webcam
so address to connect would be http://<your printer IP>/webcam?action=stream
also in my case the mjpg-streamer.sh wasn't present after installing the update so i created it manually and it started working

jbatonnet, 2024-12-07:
Yeah it's not mentionned because it's the default URL Mainsail/Fluidd use for the webcam. Using release starting from 05.12.24, it should include `mjpg-streamer` binaries and the `mjpg_streamer.sh` script.

The latter is copied form the updated update.sh script, here: https://github.com/utkabobr/DuckPro-Kobra3/pull/41/files#diff-638815ac5320cf5cffd565d2bafac2c3d08b05413a608d04434c9d7c95b8441dR103
As mentioned by other people, mjpg_streamer.sh checks if `/useremain/home/ytka/printer_data/config/.camera` exists, then checks if the default USB camera path exists `/dev/video10` (this one might vary I didn't check). It both exist, it will kill the default gkcam, and run mjpg-streamer with resolution 1280x720 instead.

The included `mjpg-streamer` binary already embeds the `input_uvc.so` and `output_http.so` plugins so no need to worry about that.

### TL;DR
- Flash the 05.12.24 release
- Create a `.camera` file in `/useremain/home/ytka/printer_data/config/` using Mainsail, Fluidd, SSH or ADB
- Make sure your camera is plugged in
- Reboot your printer
- Create a default wencam in Mainsail or Fluidd (with MJPG Streamer and the default /webcam?action=X paths)
- Your cameras should work

If it does not work, make sure all the files / scripts / logic are working on your printer using SSH or ADB

joelpr1234, 2024-12-07:
> Yeah it's not mentionned because it's the default URL Mainsail/Fluidd use for the webcam. Using release starting from 05.12.24, it should include `mjpg-streamer` binaries and the `mjpg_streamer.sh` script.
> 
> The latter is copied form the updated update.sh script, here: https://github.com/utkabobr/DuckPro-Kobra3/pull/41/files#diff-638815ac5320cf5cffd565d2bafac2c3d08b05413a608d04434c9d7c95b8441dR103 As mentioned by other people, mjpg_streamer.sh checks if `/useremain/home/ytka/printer_data/config/.camera` exists, then checks if the default USB camera path exists `/dev/video10` (this one might vary I didn't check). It both exist, it will kill the default gkcam, and run mjpg-streamer with resolution 1280x720 instead.
> 
> The included `mjpg-streamer` binary already embeds the `input_uvc.so` and `output_http.so` plugins so no need to worry about that.
> 
> ### TL;DR
> * Flash the 05.12.24 release
> * Create a `.camera` file in `/useremain/home/ytka/printer_data/config/` using Mainsail, Fluidd, SSH or ADB
> * Make sure your camera is plugged in
> * Reboot your printer
> * Create a default webcam in Mainsail or Fluidd (with MJPG Streamer and the default /webcam?action=X paths)
> * Your camera should work
> 
> If it does not work, make sure all the files / scripts / logic exist and are working as described above on your printer using SSH or ADB

Thanks for the explanation. I can see the webcam is activated by the printer(it has a blue LED), but I can't see it, no matter I do, Mainsail or web browser, I can't see it. What service I should choose, mjpeg-streamer or adaptive mjpeg-streamer? 
Also, the default address is x.x.x.x/webcam/?action=stream, not x.x.x.x/webcam?action=stream, I tried both, not working, a white Bad Gateway screen is what I get in the browser.

jbatonnet, 2024-12-07:
Okay I see I messed up my - and _ in the mjpg_streamer commit. I'm pushing a new PR now. If it still doesn't work then, I'll provide additional steps for debugging

joelpr1234, 2024-12-07:
> Okay I see I messed up my - and _ in the mjpg_streamer commit. I'm pushing a new PR now. If it still doesn't work then, I'll provide additional steps for debugging

Glad to know you found the problem, Thanks. I though was me, I'm not a coder, just a user that likes messing with stuff.

Art531, 2024-12-07:
There is no **mjpg-streamer.sh** script in this build, just create it and everything will work:
we need to create missing script file `/useremain/home/ytka/mjpg-streamer.sh`

```
if [ -e /useremain/home/ytka/printer_data/config/.camera ]; then
    if [ -e /dev/video10 ]; then
        killall gkcam
        /useremain/dist/mjpg-streamer/mjpg_streamer -i "input_uvc.so -d /dev/video10 -n -r 1280x720" -o "output_http.so -w /useremain/dist/mjpg-streamer/www"
    fi
fi
```

And of course don't forget to add empty` .camera` file in `/useremain/home/ytka/printer_data/config/`

joelpr1234, 2024-12-08:
> There is no **mjpg-streamer.sh** script in this build, just create it and everything will work: we need to create missing script file `/useremain/home/ytka/mjpg-streamer.sh`
>
> ```
> if [ -e [...]

I appreciate your effort, but unfortunately I don't know how to do that, my idea is creating at txt file in notepad, pasting your code, saving it as mjpg-streamer.txt, then changing the extension to .sh, then I'm lost, I know I need SSH and possibly chmod the file but I don't know what else I have to do to move the file and make it work.

The empty file .camera in the config I did it in mainsail, it will be great if I could navigate to other folders in the config files area of mainsail, so I could create the .sh file from there.

Art531, 2024-12-08:
> I appreciate your effort, but unfortunately I don't know how to do that, my idea is creating at txt file in notepad, pasting your code, saving it as mjpg-streamer.txt, then changing the extension to [...]

No problem, just follow these steps:
1. Use Putty to SSH to your printer using root/rockchip as login and password
2.  `cd /useremain/home/ytka`
3.  `vi mjpg-streamer.sh`, press `i` to insert text
4. Copy code from my prev. post and paste it in opened putty window
5. Press Esc, then type `wq!` and hit Enter to save and exit
6. `chmod +x mjpg-streamer.sh`
7. Connect your camera and reboot the printer, everything should work now.

joelpr1234, 2024-12-08:
> No problem, just follow these steps:
>
> 1. Use Putty to SSH to your printer using root/rockchip as login and password
> 2. `cd /useremain/home/ytka`
> 3. `vi mjpg-streamer.sh`, press `i` to insert text
> 4. [...]

Thanks. I tried and there is an error, vi to save and exit is `:wq!` but I got it looking a How To: Vi website.
I'm still getting the white Bad Gateway error in the browser, and I get this trying to configure the camera:

[image]

I don't know what else to do at this point.

Thank you again for taking the time to write up the procedure, I really appreciate it.

joelpr1234, 2024-12-08:
> @joelpr1234 i don't think it's mentioned anywhere, but if you go to nginx conf file you can see that camera is at /webcam so address to connect would be `http://<your printer [...]

@tymtymm please, could you explain me how you did it? I looked my nginx.sh file and I don't see anything  about the webcam, I used vi command to access it, not sure if that is a good way to look for that, but I used :q! to exit without saving. Thanks.

Art531, 2024-12-08:
@joelpr1234 try to remove ip address in your URL stream, like this:
[image]

joelpr1234, 2024-12-08:
> @joelpr1234 try to remove ip address in your URL stream, like this: ![изображение_2024-12-08_212030198](https://private-user-images.githubusercontent.com/60143840/393632416-05771920-5724-4db0-ab35-467f99822d65.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM2ODkzNzYsIm5iZiI6MTczMzY4OTA3NiwicGF0aCI6Ii82MDE0Mzg0MC8zOTM2MzI0MTYtMDU3NzE5MjAtNTcyNC00ZGIwLWFiMzUtNDY3Zjk5ODIyZDY1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjA4VDIwMTc1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIzYzQ1ZDYwYzEyYjBmOTc1MmY0ZWQ1YjIxMzI3NGFhNGVkNmU3YjVkOGZiZDUyZjYwZWYyY2E1MGE0NDI2YTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.K0qWetguFAH1F84UZxobDsUdNRwpnUX6B1e1EkhTLV4)

@Art531 

I feel very dumb, now it is working! Thank you all for helping me. Now I'm going to do it in my second k3 unit.

Best regards

ducati1198, 2024-12-15:
Hello after doing through this issue, I still have no camera feed try all the config possible and check all the files, everything looks ok or maybe im missing something

Art531, 2024-12-15:
Post your camera config screenshot here, also _/useremain/home/ytka/printer_data/config/_ folder content and /useremain/home/ytka/mjpg-streamer.sh file content.
What usb port you are using for camera? Anycubic says that only left one will work with camera, not right one.

ducati1198, 2024-12-15:
im using the left usbmy ngnix.conf file and everthin related to camera looks there

[image]
[image]
[image]

[image]

ducati1198, 2024-12-15:
[image]

Art531, 2024-12-15:
Can you start mjpg-streamer.sh manually? Is it executable (chmod +x mjpg-streamer.sh)?

ducati1198, 2024-12-15:
thank you my friend its work 
[image]

joelpr1234, 2024-12-25:
> thank you my friend it work ![Screenshot 2024-12-15 at 15 27 31](https://private-user-images.githubusercontent.com/76837869/395893644-34be899b-753c-48a5-af0c-0f5b4b236670.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUxNTY4MzgsIm5iZiI6MTczNTE1NjUzOCwicGF0aCI6Ii83NjgzNzg2OS8zOTU4OTM2NDQtMzRiZTg5OWItNzUzYy00OGE1LWFmMGMtMGY1YjRiMjM2NjcwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjI1VDE5NTUzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlhZTYzMGMyNDMwMTg3YzMwOTU0YjEzZWNkY2RjZjBkNzU1MzRlZTA1NTYzNjM3OTcyZDc0MmNiOWU3NzI1MjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.UfBWc0W4bj7CKL5usdxFZvqrnjM-Wb0GCz2StptSpZ4)

What camera are you using? It has a very wide-angle lens; I like it.

joelpr1234, 2024-12-25:
> Доброго времни суток коллеги, может кто сталкивался и подскажет, камеру я настроил и в вэб интерфейсе все работает (если сижу с ПК), но если зайти через мобильный браузе или с планшета или подключиться через Orca слайсер вместо камеры серый экран, как победить данную проблему? Так же не знаю в это ли обсуждение писать но заметил что если подключен через вэб интерфейс то после переключения с видеокамеры в раздел Система замечаю что процессор загружен на 100% в связи с чем (как я предполагаю) иногда принтер при начале печати творит странные вещи и/или klippper вовсе перезагружается, уже страшно подключаться во время печати вдруг все пойдет не по плану, на данный момент уменьшил разрешение до 800/600 буду наблюдать за работой принтера но возможно кто то уже сталкивался с подобным и тое как то решил проблему?
> 
> eng Good afternoon, colleagues, maybe someone has come across and can tell me, I have set up the camera and everything works in the web interface (if I'm sitting from a PC), but if you log in via a mobile browser or tablet or connect via an Orca slicer instead of a gray screen camera, how to overcome this problem? I also don't know if I should write about this discussion, but I noticed that if I am connected via the web interface, then after switching from the video camera to the System section, I notice that the processor is loaded at 100%, which is why (as I assume) sometimes the printer does strange things at the beginning of printing and/or the klipper reboots altogether, it is already scary to connect to printing time suddenly everything does not go according to plan, at the moment I have reduced the resolution to 800/600. I will monitor the printer, but maybe someone has already encountered this and somehow solved the problem?

I'd rather decrease the FPS to 5 than decrease the resolution.

AbrakadabraWS, 2024-12-25:
> What camera are you using? It has a very wide-angle lens; I like it.

Logitech HD Pro Webcam C920

Пока что со сниженным разрешением работает нормально нагрузка на процессор не более 75%
Идея со снижением FPS  тоже хорошая, конечно. же хочется получить максимально качественную и плавную картинку но видимо с этим процессором это не выйдет, подумываю поставить отдельно RaspberyPi с камерой заодно на неё снимать таймлэпс видео, кстати как на этой прошивке снимать таймлэпс если печать запускаешь через вэб интерфейс?

EN
So far, the processor load is working fine with a reduced resolution of no more than 75%
The idea of lowering FPS is also a good one, of course. I also want to get the highest quality and smoothest picture, but apparently it won't work with this processor, I'm thinking of installing RaspberyPi separately with a camera at the same time to shoot video timelaps on it, by the way, how to shoot timelaps on this firmware if you run printing through the web interface?

NatroNx, 2025-01-03:
Is there a way to add a rtsp stream camera to the Mainsail Interface? 

I tried with HLS, but without sucess.

DonatoTaucci, 2025-01-18:
> No problem, just follow these steps:
>
> 1. Use Putty to SSH to your printer using root/rockchip as login and password
> 2. `cd /useremain/home/ytka`
> 3. `vi mjpg-streamer.sh`, press `i` to insert text
> 4. [...]

@Art531 with Putty I'm trying to access by inserting root/rockchip, but it always gives me access denied. Why?

Art531, 2025-01-18:
@DonatoTaucci do not insert whole string, just follow onscreen requests - when you start Putty and connecting to some host first of all it asks you about your login, so you need to type _root_ and press enter. And after that it will ask you about your password, so type _rockchip_ and hit enter, and since it is your password - you will not see it during pressing keys, but it will be entered in console.

lequim2010, 2025-02-23:
[image]

I get this error, how can I do it?