utkabobr/DuckPro-Kobra3 #53: Cant connect to anycubic cloud (not connected)

ducati1198, 2024-12-08:
On stock and duckpro firmware anycubic account is always saing not connected. What can i do 
[image]

joelpr1234, 2024-12-11:
to be honest, I was trying to fix the camera on mainsail so I could drop the cloud, now I have the printers in LAN mode only. I had trouble to use the cloud anyway, but to connect at that point you are is going back to the app and adding the printer using the barcode scanner in the app, even if the printer is there, just add it again, it will fix the problem.

ducati1198, 2024-12-11:
I did try adding the printer back but doesn’t work. Try manually end with the Bluetooth with no success.

utkabobr, 2024-12-11:
It seems that your device.ini may be messed up in some way, try to check /userdata/app/gk/config/device.ini. Cloud services depends on it

ducati1198, 2024-12-11:
Hello this is what is in devive.ini
[image]
[image]

ducati1198, 2024-12-11:
I think i found fille that are missing in useremain.
Im missing the useremain/app/gk/cert3 and cert2 and one. See pictures 
[image]
[image]

ducati1198, 2024-12-11:
After verifying im missing cert1 and 3 folder bit im mot sure if there’s other folder or files missing in my useremain folder, can someone send me a copie of that folder for me to compare, please

joelpr1234, 2024-12-11:
How they got missing?

ducati1198, 2024-12-11:
I deleted the folder without doing by purpose. 😔

joelpr1234, 2024-12-11:
ok, glad to know you fix it.