utkabobr/DuckPro-Kobra3 #54: Spool Macros Not working. Any help?

ducati1198, 2024-12-09:
Hello quick question when  spoolman  in moonracker it show up  in mainsail i cam chose the filament and eject it with mainsail. Now the issue is that when adding spoolman macros to printer.cfg klipper crashes and doesn’t retard. Mainsail and moonracker still work, removing the macros in printer.cfg klipper work again. Any help with this, thanks you

Printer: Kobra 3 combo
Printer Version: latest
Ace pro version: lastest

moonraker.conf: 

[spoolman]
server: http://192.168.0.123:7912
#   URL to the Spoolman instance. This parameter must be provided.
sync_rate: 5
#   The interval, in seconds, between sync requests with the
#   Spoolman server.  The default is 5.

Printer.cfg:

[gcode_macro SET_ACTIVE_SPOOL]
gcode:
  {% if params.ID %}
    {% set id = params.ID|int %}
    {action_call_remote_method(
       "spoolman_set_active_spool",
       spool_id=id
    )}
  {% else %}
    {action_respond_info("Parameter 'ID' is required")}
  {% endif %}

[gcode_macro CLEAR_ACTIVE_SPOOL]
gcode:
  {action_call_remote_method(
    "spoolman_set_active_spool",
    spool_id=None
  )}
[image]
[image]
[image]

utkabobr, 2024-12-11:
I don't think that goklipper supports action_call_remote_method

ducati1198, 2024-12-11:
can you hat the big difference on goklipper and klipper like in the creality k1. i try looking on google for what is goklipper but i can find info on it. i been printer on a fully mod crealitu k1 max and just got this kobra 3 and looking forward to have the same feature like the creality. BTW thank for your work on this root. i love having klipper rooted :

DrummerKH, 2024-12-11:
@ducati1198 Hello, where did you get the ACE Pro's Ip address and port from?

utkabobr, 2024-12-11:
> can you hat the big difference on goklipper and klipper like in the creality k1. i try looking on google for what is goklipper but i can find info on it. i been printer on a fully mod crealitu k1 [...]

https://github.com/ANYCUBIC-3D/Kobra3/tree/main/klipper-go

It's completely different language, it definitely changed in multiple ways, so action_call_remote_method might be just missing. It can not be fixed as source code of goklipper is not full

ducati1198, 2024-12-11:
> @ducati1198 Hello, where did you get the ACE Pro's Ip address and port from?

# moonraker.conf

[spoolman]
server: http://192.168.0.123:7912
#   URL to the Spoolman instance. This parameter must be provided.
sync_rate: 5
#   The interval, in seconds, between sync requests with the
#   Spoolman server.  The default is 5.

ducati1198, 2024-12-11:
> https://github.com/ANYCUBIC-3D/Kobra3/tree/main/klipper-go
>
> It's completely different language, it definitely changed in multiple ways, so action_call_remote_method might be just missing. It can not [...]

Hoo I understand know let all thanks Anycubic for saying it’s will be open source 😂. In dockers open source mine’s FULL OPEN

utkabobr, 2024-12-11:
> > # moonraker.conf
> > [spoolman] server: http://192.168.0.123:7912
> 
> The moonraker tells - this server is not available. So i guess it is different for some printers. Where did you get this address?

Definitely. You must setup your own server of spoolman

ducati1198, 2024-12-11:
You need to install spoolman on a server. Setup spoolman after that configuration moonraker on your printer. It would be amazing if we could make spoon and work on the kobra 3. Superman has been great with my creality k1m to keep track of my filaments

kiinami, 2024-12-12:
I managed to make it work on my Kobra 3, in DuckPro version [05.10.24-02](https://github.com/utkabobr/DuckPro-Kobra3/releases/tag/05.10.24-02), a month ago, after much toiling and printer rebooting. I have not yet updated in fear it breaks so I can't tell you if it works with the most recent version.

## TL;DR:
1. Get yourself a configured Spoolman instance somewhere the printer can reach (network-wise). Follow [their wiki](https://github.com/Donkie/Spoolman/wiki/Installation).
2. In `printer.cfg` add:
   ```ini
   [gcode_macro SET_SPOOL]
   gcode:
       {{ action_call_remote_method("spoolman_set_active_spool", params) }}

   [gcode_macro M555]
   gcode:
       SET_SPOOL SPOOL_ID={{ params.S }}
   ```
3. In `moonraker.conf`, add:
   ```ini
   [spoolman]
   server: http://<spoolman ip>:<spoolman port>
   sync_rate: 5
   ```
4. SSH inside the printer and get to `/useremain/home/ytka/moonraker/moonraker/components/spoolman.py`. Change the following lines:
   ```diff
   diff --git a/moonraker/components/spoolman.py b/moonraker/components/spoolman.py
   index d8a3e1d..51728c1 100644
   --- a/moonraker/components/spoolman.py
   +++ b/moonraker/components/spoolman.py
   @@ -277,7 +277,9 @@ class SpoolManager:
            else:
                self.pending_reports[spool_id] = used_length
    
   -    def set_active_spool(self, spool_id: Union[int, None]) -> None:
   +    def set_active_spool(self, spool_id: Union[int, None] = None, SPOOL_ID: Union[int, None] = None) -> None:
   +        if spool_id is None:
   +            spool_id = int(SPOOL_ID)
            assert spool_id is None or isinstance(spool_id, int)
            if self.spool_id == spool_id:
                logging.info(f"Spool ID already set to: {spool_id}")
   
   ```
5. End the SSH connection and reboot after like 5 minutes because sometimes the files don't get written if you reboot immediately
6. Reboot your printer
7. In your slicer of choice, add in the filament start code `M555 S=<spool id>` in lieu of `SET_ACTIVE_SPOOL ID=<spool id>`

## Backstory

As said before, Kobra 3 runs on Goklipper, which frankly is a mess when touching `printer.cfg` or trying to add any custom config. Since it runs in Goklipper, the template engine that parses the `printer.cfg` file is not jinja2 but a port called [pongo2](https://github.com/flosch/pongo2). This means that all the stuff inside the `{{ }}` must be written in Go, not in Python.

That in itself is easy, as the translation is quite straightforward, except for a very specific issue. You see, goklipper's `action_call_remote_method` needs a Go object to be passed with the parameters that will then be passed to the remote method itself, which in this case is the spool ID. There is, however, no way to create a Go object inside the template engine. Or at least none I could find after five days of bashing my head against the problem.

The solution I found was to pass the entire `params` object, which is thankfully available and includes the spool ID argument. All arguments keys (e.g. SPOOL_ID) are capitalized by goklipper even if you write them lowercase in `printer.cfg`.

That goes directly to `moonraker/components/spoolman.py`, where it is passed as keyword arguments to the function. Since keyword arguments are case-sensitive in Python, and goklipper uppercases the argument keys, the fix in step 4 is needed for it to be correctly passed to the function. There is probably a more elegant way of doing it, but I was desperate at that point.

With that, the command works in console, but strangely it did not work from gcode files, which is where spool changes usually happen. After some trial and error, I discovered goklipper will only execute commands inside gcode files if they follow the format `M###` and whose parameters keys are only one character long. Thus, we arrive to the final `M555 S=<spool id>` command that acts as a proxy to the `SET_SPOOL SPOOL_ID=<spool id>` and that goklipper will execute. Those two could probably be merged, as long as the added parameter in `moonraker/components/spoolman.py` is changed to `S` instead of `SPOOL_ID`

Hope this helps, and if it works for everyone, maybe it could be merged. Thanks to @utkabobr for his amazing work on this project :)

ducati1198, 2024-12-13:
Wow that great information. Will try it on the latest version soon. I will report back

utkabobr, 2024-12-13:
> With that, the command works in console, but strangely it did not work from gcode files, which is where spool changes usually happen. After some trial and error, I discovered goklipper will only [...]

WOW! THAT'S WHY BED_MESH_CALIBRATE FAILED TO WORK.
I've been struggling to figure out why it works only from console to remove that limitation.
Thanks for the great information, will try to patch gklib binary to bypass that.

ducati1198, 2024-12-13:
> ```diff
> diff --git a/moonraker
> ```

Can you elaborate in moonraker.py what lines to be replaced with the lines mentioned above thank you

ducati1198, 2024-12-13:
> ```diff
> self.pending_reports[spool_id] = used_length
> ```

Could it be we need to change this 2 line (original moonraker.py) to the line about you mentioned 
[image]
[image]

ducati1198, 2024-12-13:
i try on the lastest version with no success. im probeble doing something wrong

moonracker log error:

/useremain/home/ytka/moonraker/moonraker/components/spoolman.py", line 275
    index d8a3e1d..51728c1 100644
                       ^
SyntaxError: invalid decimal literal
1970-01-01 08:00:13,500 [server.py:add_warning()] - Unparsed config section [spoolman] detected.  This may be the result of a component that failed to load.  In the future this will result in a startup error.

and pictures of what i change in moonraker.py
[image]

kiinami, 2024-12-13:
> i try on the lastest version with no success. im probeble doing something wrong
>
> moonracker log error:
>
> /useremain/home/ytka/moonraker/moonraker/components/spoolman.py", line 275 index [...]

Don't just copy-paste the entire code block. That is a diff summary, it expresses the lines that are changed and where they need to be changed. The red lines are the lines that have to be deleted, and the green lines are the lines that have to be added instead. Checkout [this guide](https://www.atlassian.com/git/tutorials/saving-changes/git-diff) to understand how `git diff` works

ducati1198, 2024-12-13:
AMAZING I GOT IT WORKING. THANK YOU VERY MUCH. WHAT DO YOU THINK ABOUT Toolchanger Support in fluidd can it work?

ducati1198, 2024-12-13:
I can confirm that its works on version 2.3.5.3. I cant this with the printer because my hotend is broken, im getting my new one on moday and will test

ducati1198, 2024-12-15:
After getting my new hotend and test a multi color print i have these results: switching between filament is working, but i get no reading filament use to spoolman server. The filament remaining stays the same. I try its with orca slicer and orca anycubic and have the same results.

Bespinuk, 2024-12-15:
ive just tested this and it appears to be working with the one print that i have done.

NatroNx, 2025-01-02:
> I managed to make it work on my Kobra 3, in DuckPro version [05.10.24-02](https://github.com/utkabobr/DuckPro-Kobra3/releases/tag/05.10.24-02), a month ago, after much toiling and printer rebooting. [...]

this worked (almost) perfect for me, but maybe you can help me to improve.

I found that once I use the 555 macro it seems to use the ID of the Filament in my slicer, but one less?

I tried to use Filament 3 from NextSlicer but switched to FIlament 2. Maybe the ordering starts at 0 somewhere?

[image]

kiinami, 2025-01-03:
> this worked (almost) perfect for me, but maybe you can help me to improve.
>
> I found that once I use the 555 macro it seems to use the ID of the Filament in my slicer, but one less?
>
> I tried to use [...]

It is the Spoolman Spool ID, that you can get in the first column of the Spools page table. The numbers in the slicer have nothing to do, they just denote order.

NatroNx, 2025-01-04:
I apologize. I didn't understand whats happening. Had a look now at my gcode and I'm confused.

I thought the slicer would, depending on those numbers, fill the variables in the gcode and so I'd send M555 S=1 (for example) for the first color in the slicer.  But this is not the case as I now know. The slicer sends M555=<spool id> and not changing that variable and for some reason I don't yet understand klipper itself changed this. 

But to what? Which spool id would be set if I choose the very left filament in Ace Pro?

ducati1198, 2025-01-04:
I dont know if you guys have the same problem as me, I always need to reboot spoolman app in my server docker to be able to read the filament count.

utkabobr, 2025-01-20:
> WOW! THAT'S WHY BED_MESH_CALIBRATE FAILED TO WORK. I've been struggling to figure out why it works only from console to remove that limitation. Thanks for the great information, will try to patch [...]

Small update - got my hands on it, unlocked all the gcode commands. Will make a universal patcher sometime later.
Just replace your /userdata/app/gk/gklib with a new one, don't forget to modify permissions via `chmod +x /userdata/app/gk/gklib`

[gklib.zip](https://github.com/user-attachments/files/18482778/gklib.zip)

ducati1198, 2025-02-11:
> Small update - got my hands on it, unlocked all the gcode commands. Will make a universal patcher sometime later. Just replace your /userdata/app/gk/gklib with a new one, don't forget to modify [...]

Thank you. Do i need to put back the original text in spoolman.py to work with your patch?

utkabobr, 2025-02-11:
> Thank you. Do i need to put back the original text in spoolman.py to work with your patch?

It only unlocks commands, so no need to put it back I suppose

nolantern, 2025-02-26:
> Small update - got my hands on it, unlocked all the gcode commands. Will make a universal patcher sometime later. Just replace your /userdata/app/gk/gklib with a new one, don't forget to modify [...]

Hey does this gklib version also restore the BED_MESH_CALIBRATE behaviour of original klipper?