utkabobr/DuckPro-Kobra3 #55: Non official firmware code has  been detected, which damage  the printer. Please use it with  caution.

Isser-yts, 2024-12-12:
Kinda error (notice) which I have after installation DuckPro. I guess it is bootloop I have with white fully fulled line.

Предупреждение о неофициальной прошивке и дальнейшая вечная загрузка с заполненной белой полосой.

Почему так?

brestows, 2024-12-20:
I have the same problem

nolantern, 2024-12-20:
Same here.

### Here are the exact steps I have done:

1. On FW 2.3.2.9 I successfully installed DuckPro [05.10.24](https://github.com/utkabobr/DuckPro-Kobra3/releases/tag/05.10.24)
2. When FW 2.3.5.3 was available installed it via ota but lost DuckPro. Flashing release 05.10.24 was not successful anymore.
3. When releases [03.12.24](https://github.com/utkabobr/DuckPro-Kobra3/releases/tag/03.12.24) and [05.12.24](https://github.com/utkabobr/DuckPro-Kobra3/releases/tag/05.12.24) where released I flashed them on top of FW 2.3.5.3. After reboot I got the message "Non official firmware code has been detected, which damage the printer. Please use it with caution." on the display. When clicking "OK" the message dissappears but the bootscreen with a full (or empty? I don't know anymore) progressbar stayed forever* (*=serval hours). Reboots resulted here again and again.
4. I also tried to flash [Mutable config cleanup script](https://github.com/utkabobr/DuckPro-Kobra3/releases/tag/cleanup_mutable_config) (after flashing 03.12.24 or 05.12.24) but it happened nothing after inserting the USB stick. Even after hours the update.swu was available on the flashdrive.
5. I successfully flashed FW 2.3.5.3 to use my printer without DuckPro since then.

jbatonnet, 2024-12-20:
The non official code warning has nothing to do with the issues you have with the printer / firmware. If the printer is detecting that SSH, ADB or nginx (Mainsail / Fluidd) is running, it will show this message. But that's it.
Some work has been done [here by systemik](https://github.com/systemik/Kobra3-Firmware/commit/ed32e76c7148c331461cb786986e157053b9fec7#diff-f5913def28bae4b47faf63106a84b3f1f2ca7f2222ce9c6ad07fab1855bb26e1) and [here by me](https://github.com/jbatonnet/Rinkhals/issues/3) to disable this screen but t is purely visual.

For the main issue (bootloop, white line), you should try to performe a completely clean installation, downgrading to 2.3.3.9, factory reset then reapplying 2.3.5.3 and then custom firmware.
I wrote some documentation on my own firmware repo about how to recover from another error but it would be the same process: https://github.com/jbatonnet/Rinkhals/wiki/Firmware#my-printer-shows-a-11407-error

tommmmmd, 2024-12-29:
du kannst auch mit filezilla auf den Drucker zugreifen und das installierte verzeichnis von Duckpro komplett manuell löschen und anschl. neu flashen, so hat es bei mir funktioniert.

Simfony-star, 2025-01-05:
> du kannst auch mit filezilla auf den Drucker zugreifen und das installierte verzeichnis von Duckpro komplett manuell löschen und anschl. neu flashen, so hat es bei mir funktioniert.

Can you tell me which directories should be deleted?

ducati1198, 2025-01-06:
What I recommend is to flash the factory firmware. After use rinkhals (github) ssh tool this will give you back access to ssh on port 2222. Ssh in and cd  /useremain after rm -rf home (it will delete duckpro folder). After reboot and reflash duckpro. You can always try rinkhals root for kobra 3