utkabobr/DuckPro-Kobra3 #58: Mobileraker companion

Bespinuk, 2024-12-15:
Looking for a way to install the mobileraker companion, but im not sure how you build into the firmware.

Bespinuk, 2024-12-22:
surprised no one is using the mobileraker app as it would give us a way of monitoring the printer other than the anycubic app if it is working.

ducati1198, 2024-12-22:
You can use octoapp, it work 
[image]

Bespinuk, 2024-12-22:
Thank you i will try that, was thinking mobile raker as it as spoolman integrated. Also would be great if we could get obico working then we might get good AI detection.

ducati1198, 2024-12-22:
Octoapp as spoolman support

[image]

Bespinuk, 2024-12-22:
Excellent I might switch all my printers to this as it as watch support too. I had not used it in years as was marlin.

NatroNx, 2024-12-31:
> You can use octoapp, it work ![IMG_6432](https://private-user-images.githubusercontent.com/76837869/398016505-97f8675e-82e9-4eee-b222-039009ef0f68.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU2MDUzNDMsIm5iZiI6MTczNTYwNTA0MywicGF0aCI6Ii83NjgzNzg2OS8zOTgwMTY1MDUtOTdmODY3NWUtODJlOS00ZWVlLWIyMjItMDM5MDA5ZWYwZjY4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjMxVDAwMzA0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMxNGJkZTRkODZjYTk0ZDQ1MGQ3Y2MwZDJmMDU5ZGYyMjQxNzkxOGU0OTU5YmYyYjhmZWY5MjUwMjQ4NWQyZTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.nKq8pOq_V97i6DL6TEuSYDbXel7aEH_Zb9bkzZ0YVds)

Octoapp on Iphone connects but fails and reconnects every few seconds

Octoapp on Android doesn't even connect for me