utkabobr/DuckPro-Kobra3 #59: code 10131

jeanxxl13, 2024-12-19:
Hi!
After trying to install the firmware, the error code 10131 started to appear on the screen, after, I tried to install versions (2.3.5.3/2.3.3.9) the error still does not go away. I checked all the cables, did not find any problems

J0shweb, 2024-12-21:
Try using the file from this https://wiki.anycubic.com/en/error-codes/10408-code/k3 if you have re-flashed firmware multiple times your tool-head and mobo firmware's might be too far apart.

utkabobr, 2024-12-22:
10131 may indicate that your MCU-s have different firmware versions because you have tried to lower your firmware.
It will be very tricky to re-flash after it, you need to setup gklib with firmware for one of them (matching MCU version), boot almost empty printer.cfg and flash new firmware (Check info in #28)

jeanxxl13, 2024-12-23:
Could you help me solve this problem? connecting via anydesk. I will pay you for your help. Thank you

J0shweb, 2024-12-23:
> Could you help me solve this problem? connecting via anydesk. I will pay you for your help. Thank you

I can help you with creating the USB from the link i have provided. Reply if you would like me to assist you with that. (i won't charge you any money)

jeanxxl13, 2024-12-23:
I was able to make a USB from the link, but it didn't help. It must be done according to the utkabobr method

jeanxxl13, 2024-12-23:
> > Could you help me solve this problem? connecting via anydesk. I will pay you for your help. Thank you
> 
> I can help you with creating the USB from the link i have provided. Reply if you would like me to assist you with that. (i won't charge you any money)

Let's try

jeanxxl13, 2024-12-23:
> > Could you help me solve this problem? connecting via anydesk. I will pay you for your help. Thank you
> 
> I can help you with creating the USB from the link i have provided. Reply if you would like me to assist you with that. (i won't charge you any money)

My Discord Profile :  Roman4212

Dovhun, 2025-01-24:
> I was able to make a USB from the link, but it didn't help. It must be done according to the utkabobr method

I have the same problem, did you solve it?

ducati1198, 2025-01-24:
Try rinkhals kobra 3 root. Its will probably work and you get screen printer working too 

https://github.com/jbatonnet/Rinkhals

jeanxxl13, 2025-01-24:
> > I was able to make a USB from the link, but it didn't help. It must be done according to the utkabobr method
> 
> I have the same problem, did you solve it?

It worked out according to this link, you just have to wait longer
 https://wiki.anycubic.com/en/error-codes/10408-code/k3