utkabobr/DuckPro-Kobra3 #60: Missing sensors using Moonraker integration in home assistant

Hinguy, 2024-12-20:
Hello,
I got the moonraker custom integration in home assistant working with the ducker cfw on my Kobra 3. It is showing all the buttons to control the printer, however all the sensors are missing, so I can't integrate them into home assistant to do automations. Is anyone also using home assistant with the duck pro firmware and got this working?

Thanks for the help, well appreciated! 👍

Bespinuk, 2024-12-20:
I suspect that is because it is taking the information from the cloud integration and not from the printer, and the printer is not sending the same information to the amycubic cloud

Hinguy, 2024-12-30:
> Hello,
> 
> how did you connect your Moonracker connected to homeassistant? Which port did you use?

I just used the Custom Integration from here:
https://www.home-assistant.io/docs/quality_scale/#-custom

For the Port I used the default Port for moonraker, which is 7125. I Hope This helps.

NatroNx, 2024-12-30:
Ok got it - did the same with the same outcome. Lots of Macros and Controls but no sensors