utkabobr/DuckPro-Kobra3 #61: Printer stuck in the loading screen

emrexx13, 2024-12-22:
Hi I tried to download this firmware. I did it then the printer reboot and open again loading bar getting full multiple times and I saw the notification about it's non official firmware I press okay but the loading screen stuck. The printer doesn't open and staying this screen for hours. I also power off and on again but nothing change. Can anyone help me about how can I fix this problem?

ducati1198, 2024-12-22:
You need reflash duckpro 2.3.5.3

emrexx13, 2024-12-22:
Thank you for your replying. I will do it again. Have a good day.

Android için Outlook<https://aka.ms/AAb9ysg> edinin

derrene666, 2025-01-01:
So your bar reloads all the time or not?

If no:
If you get into mainsail, delete all printer mutable and restart the printer.

anselor, 2025-01-14:
I have this same problem. Printer stuck on the loading screen. Initially the progress bar is moving as it boots and then the progress bar resets to no progress. Printer doesn't get on WiFi so can't connect to it.

anselor, 2025-01-14:
Any help would be appreciated. Printer is otherwise bricked.

anselor, 2025-01-14:
Can't flash the stock firmware either. No response to plugging in the USB drive with stock firmware on it.

Gorkembukel, 2025-01-17:
Same thing here. Anything new?

ducati1198, 2025-01-17:
Did you change the name to update.swg on your usb

ducati1198, 2025-01-17:
If its don’t work you can try rinkhals root to see if you can get it working

Gorkembukel, 2025-01-17:
Although I'm not sure what changed, I formatted my USB drive, re-uploaded the folders, and tried again. This time, it worked successfully.

anselor, 2025-01-17:
Didn't get a chance to update on this. Changing the file name let me flash back to stock and the second try let me apply the custom changes. I was able to use the mesh visualization and aluminum tape to get my bed level finally. 

Was looking forward to printing with Orca but the complete disconnect between printing with DuckPro and the device screen plus the inability to select filament mappings when printing kind of makes this not super useful.

8strks, 2025-01-20:
I have tried most of the troubleshooting suggestions and successful flashed the FW Duck pro 2.3.5.3. 

1. Format your USB, *I used W11 format FAT32 drive.
2. Be sure to have aGVscF9zb3Nf folder inside the USB, update.swu inside the aGVscF9zb3Nf folder.
3. Eject the drive properly 
4. Double Check your current FW before flashing!! should be  2.3.5.3.1.2.4 ( always have WIFI connection set in the printer when flashing any FW)
5. Flash the FW and wait till it reboot and you will get the loading screen again. 
6. You will now be able to connect to your printer by using your browser 
7. Put it the IP in and boom you're in MoonSail, Find the mutual files and preconfig( maybe you don't have to for this file). Delete it from the machine and then press Reboot button (top right side ).
8. Restart your Printer after 2 mins
9. This should boot into the duckpro FW , check your printer info!