utkabobr/DuckPro-Kobra3 #66: jj

rabbit3dcustom, 2024-12-29:
jj

utkabobr, 2024-12-29:
LCD works via anycubic's bridge to Klipper, it's not supported yet

ducati1198, 2024-12-29:
If you want screen print check rinkhals root but as others little bugs.

rabbit3dcustom, 2024-12-29:
Thanks a lot both! 

> LCD works via anycubic's bridge to Klipper, it's not supported yet
@utkabobr  do you have any idea when it could be supported? 

and what happened with the purge menu?

I'm considering then to downgrade to the original firmware and continue using the USB stick till i destroy it for being plug and unplug millions of times...

But having and eye on this cfw and all the news you bring to us

Thanks a lot again
R3dC

utkabobr, 2024-12-29:
I'm thinking about proxy to anycubic's gkapi, so moonraker can be main interface for klipper and K3SysUi can continue to work. No ETA as always

rabbit3dcustom, 2024-12-29:
> I'm thinking about proxy to anycubic's gkapi, so moonraker can be main interface for klipper and K3SysUi can continue to work. No ETA as always

Thanks a lot! so appreciate! 

I will take a back seat and stays here!
Rabbit

rabbit3dcustom, 2024-12-29:
@utkabobr one last question, 

To return to the stock firmware: "Place it as update.swu in that folder with strange symbols." 

Which folder do you mean?

The one that is inside your cfw o the other one with Chinese simbols?

[image]

Apologize me but for me both has estrange symbols... lol
Rabbit

utkabobr, 2024-12-29:
First one

jbatonnet, 2024-12-31:
> I'm thinking about proxy to anycubic's gkapi, so moonraker can be main interface for klipper and K3SysUi can continue to work. No ETA as always

I've done it here if you want: https://github.com/jbatonnet/Rinkhals/blob/master/files/4-rinkhals/usr/share/scripts/moonraker-proxy.py
Prints from Slicers (octoprint compat) and Web (websocket) are supported and show up on the screen.

utkabobr, 2024-12-31:
> I've done it here if you want: https://github.com/jbatonnet/Rinkhals/blob/master/files/4-rinkhals/usr/share/scripts/moonraker-proxy.py Prints from Slicers (octoprint compat) and Web (websocket) are [...]

I suppose it's converting moonraker calls with gkapi calls, but I think it's better to use reverse approach, where we only show print progress to the K3SysUi (Change gkapi's http port to another one and send events to K3SysUi from moonraker)
It will theoretically allow using K3SysUi if/when anyone ports vanilla Klipper to Kobra-s

jbatonnet, 2024-12-31:
Sure, those are two different approaches and I fully respect that. What you describe would be the best path for a fully open printer, and I would even say that ideally we should have a reimplementation of the printer interface someday (it's a simple visual app using the framebuffer after all).
My goal is more functional, I want to use the printer with its current features and apps but I'm trying to make it work in an open ecosystem.
That's cool to see different goals for different projects, the users will have more choice.
I'll follow your work then, good luck :)

utkabobr, 2024-12-31:
> Sure, those are two different approaches and I fully respect that. What you describe would be the best path for a fully open printer, and I would even say that ideally we should have a [...]

Yeah, someone are using Anycubic's services, but Obico/Octoeverywhere/Other will be better because of universal approach and more support imho