utkabobr/DuckPro-Kobra3 #67: How to safe z offset for all the printes??

ducati1198, 2024-12-30:
Is it possible to safe the z offset i set on every print i start. It look like it always default to the anycubic auto levelling. How can I permanently save the Z offset?

utkabobr, 2024-12-30:
I suppose we can't. They hardcoded many things into goklipper

NatroNx, 2024-12-31:
> Is it possible to safe the z offset i set on every print i start. It look like it always default to the anycubic auto levelling. How can I permanently save the Z offset?

If you use Anycubic Slicer Next you can define it in the machine settings

ducati1198, 2024-12-31:
> > Is it possible to safe the z offset i set on every print i start. It look like it always default to the anycubic auto levelling. How can I permanently save the Z offset?
> 
> If you use Anycubic Slicer Next you can define it in the machine settings

That the problem i don’t use it

utkabobr, 2024-12-31:
> > Is it possible to safe the z offset i set on every print i start. It look like it always default to the anycubic auto levelling. How can I permanently save the Z offset?
> 
> If you use Anycubic Slicer Next you can define it in the machine settings

It's only an offset on top of auto z-offset

derrene666, 2025-01-01:
Just dont touch the z offset or the confirm/ok button on the printer and set z offset in slicer. 547 hrs and not have to adjust the z offset. perfect first layers all the time.
slow first layer down and print first layer hotter.

ducati1198, 2025-01-01:
Where exactly you put the z offset in orca slicer pls

NatroNx, 2025-01-01:
in the prepare tab in very top left ther e is Printer. Right next to it is a edit icon - and there you can change your z-offset

derrene666, 2025-01-02:
> Where exactly you put the z offset in orca slicer pls

[image]

joelpr1234, 2025-01-03:
@derrene666 

> Just dont touch the z offset or the confirm/ok button on the printer and set z offset in slicer. 

How I do this? 

547 hrs and not have to adjust the z offset. perfect first layers all the time. (-0,02mm set in slicer. once set always stays there) slow first layer down and print first layer hotter.

Impressive. I want to try this, I can't get a good first layer and the X belt broke with a few hours of use...

Thanks

derrene666, 2025-01-03:
@joelpr1234 

First of all: I installed silicone feet instead of the sleeves.

I did:
1. Stock Fw factory reset
2. all calibrations
3. duckpro from 3.12.24
4. duckpro from 5.12.24

(it won't accept the update from 5.12 alone, no chance)

5. Changed the bed mesh and the tolerances in the cfg. Also the temperature of the bed and the nozzle at which the printer performs the ABL and the z probe

6. ABL and the bed then aligned straight (simply screwed it further in or out, the feet compensate for this very well)
6.1 all other calibrations

7. When it was straight, I printed a 1 layer and put some painters tape underneath everywhere I could see through. I was able to remove a small corner with -0.02mm z offset.

I can then send a picture of the bed with painters tape, it will print for another 14 hours.

8. Haven't done any ABL since then and haven't made any changes in the slicer either.

derrene666, 2025-01-04:
@joelpr1234 here this feet
[image]

joelpr1234, 2025-01-04:
> @joelpr1234
>
> First of all: I installed silicone feet instead of the sleeves.
>
> I did:
>
> 1. Stock Fw factory reset
> 2. all calibrations
> 3. duckpro from 3.12.24
> 4. duckpro from 5.12.24
>
> (it won't accept [...]

Hi @derrene666 

I appreciate your explanation. Unfortunately, I'm not that good with this information. If you can help me a little with step 5 in the changes to the CFG file, specifically what you changed there, I can proceed.

In my part, I changed the grid from 5x5 to 10x10. I have a weird bed in one of my two K3s, but both are about a 0.5 mm range.

[image]

[image]

I will start looking for bed silicone feet. Thanks for the picture; I thought you were talking about the printer's feet.

Thanks.

derrene666, 2025-01-04:
@joelpr1234 

I will take a look on all the changes and will post them here, maybe I can still do it today :)