utkabobr/DuckPro-Kobra3 #69: Can't get past initial boot screen

kwe1009, 2025-01-03:
I installed the latest DuckPro and after the reboot the printer is stuck on the initial boot screen.

The printer isn't getting an IP address and I can't see it on the app or from the Anycubic slicer.  I've tried to reinstall stock but no love there either.  Any suggestions?

[image]

loemic87, 2025-01-04:
Had the same problem like you. 
After that i disconnected the ACE and the camera. 
Then use the left usb-port and re-flash stock firmware. 
After that use the left usb-port and re-flash cfw.

That way worked for me.

kwe1009, 2025-01-04:
> Had the same problem like you. After that i disconnected the ACE and the camera. Then use the left usb-port and re-flash stock firmware. After that use the left usb-port and re-flash cfw.
> 
> That way worked for me.

Thanks, I'll give it a try.  Did you have the usb inserted before powering on?

ducati1198, 2025-01-04:
> > Had the same problem like you. After that i disconnected the ACE and the camera. Then use the left usb-port and re-flash stock firmware. After that use the left usb-port and re-flash cfw.
> > That way worked for me.
> 
> Thanks, I'll give it a try. Did you have the usb inserted before powering on?

No you power the printer and after insert the usb

kwe1009, 2025-01-04:
> No you power the printer and after insert the usb

That's what I did.  It's been about 45 minutes and no sounds or screen changes.  I'll let it cook for a while before doing anything else.  Thanks for the quick reply.

kwe1009, 2025-01-04:
Have the 2.3.5.3 firmware on the stick and after an hour nothing has happened.

kwe1009, 2025-01-05:
I forgot to change to the firmware name to "upgrade.swu".  Worked great after that!

closed 2025-01-05 by kwe1009