utkabobr/DuckPro-Kobra3 #72: CANCEL_PRINT macro not found in configuration.

loemic87, 2025-01-08:
As written in the title:

I have an error message in fluidd:

CANCEL_PRINT macro not found in configuration.

Is this a known issue or do i have to change anything?

sharpmonk, 2025-01-10:
Copy this code and add it into the bottom of printer.cfg then save and restart

[gcode_macro CANCEL_PRINT]
description: Cancel the actual running print
rename_existing: CANCEL_PRINT_BASE
gcode:
  TURN_OFF_HEATERS
  CANCEL_PRINT_BASE

loemic87, 2025-01-14:
@sharpmonk this did not work for me. Adding these lines at the bottom of the printer.cfg unfortunately caused the printer to stay in a boot-loop. For me it's not a problem when there is this error message, i don't face any problems with it (or i don't recognize them).

utkabobr, 2025-01-14:
It will not work. Goklipper doesn't report any info about macros to the moonraker, DuckPro's moonraker emulates such macros to suppress Mainsail warnings, I'll modify it for Fluidd later