utkabobr/DuckPro-Kobra3 #78: Is it possible to add simply print to this configuration?

DonatoTaucci, 2025-01-20:
I'm trying to add simply print to my installation, so that it's easier to access mainsail without the need for a vpn. Has anyone managed to do this or know how I could do it? I tried adding the string [simplyprint] to the moonraker.conf file, but the notification with the configuration code is not sent

vinny1575, 2025-02-10:
Im looking to do the same.

jbatonnet, 2025-02-18:
Hi, Simplyprint needs a couple of modifications for it to work
- HTTPS needs to work with proper certificates
- IPv6 needs to be deprioritized for getaddrinfo
- A small tweak in simplyprint.py

With some help, I managed to get it working on my CFW over there: https://github.com/jbatonnet/Rinkhals

And the list of commits needed to get it working for this repo: https://github.com/jbatonnet/Rinkhals/compare/20250210_01...20250217_01