utkabobr/DuckPro-Kobra3 #79: Firmware hangs on boot

Joerg1309, 2025-01-22:
I have flashed the system with the firmware. Yesterday everything ran without any problems. 
Today, after switching on, it boots through once and a message appears that it has a non-official firmware, 
which I confirm. 
After that it hangs on the screen. And nothing happens.
Too bad, that actually looked quite good. But you can't work with it like that. 

What can I do ?

[image]

Joerg1309, 2025-01-22:
After Reinstall the Original Firmware and install the Duck Pro Firmware again , the failure is the same. They not booting.

Joerg1309, 2025-01-22:
this is what I see when I enter the Ip address in the web browser

[image]

[image]

Ov1diy, 2025-01-23:
I have encountered the same issue. I've tried various solutions, but nothing seems to resolve the problem.

Guilleas21, 2025-01-23:
se queda trabado en la pantalla de carga en la pantalla ,cuanto tiempo es el normal para esperar?

Joerg1309, 2025-01-23:
> se queda trabado en la pantalla de carga en la pantalla ,cuanto tiempo es el normal para esperar?

He esperado 1 hora.  No hay progreso de carga

Guilleas21, 2025-01-23:
Si ,lo deje toda la noche y no avanzo ,apague y prendí ,pasa lo mismo ,tampoco avanza

sporterhome, 2025-01-30:
How did you flash the original firmware?  I've go the same hanging issue?  I copied the original firmware to the folder on the USB but nothing happens when I reboot

Joerg1309, 2025-01-30:
> How did you flash the original firmware? I've go the same hanging issue? I copied the original firmware to the folder on the USB but nothing happens when I reboot

Save the original firmware as update.swu in the folder. Switch on printer, plug in USB, wait, it beeps. then wait until it beeps twice, then the firmware is installed. then switch off printer.

I got it to work. If it hangs while booting you can access the printer via Ip. There I deleted all files starting with printer_mutable under machine. Then switch off and wait 5 minutes. Switch on again, then it worked.

quinman24, 2025-02-14:
> Save the original firmware as update.swu in the folder. Switch on printer, plug in USB, wait, it beeps. then wait until it beeps twice, then the firmware is installed. then switch off printer.
>
> I [...]

how dose one find the ip and port to connect and delete printer_mutable under machine?

Joerg1309, 2025-02-14:
> 
> how dose one find the ip and port to connect and delete printer_mutable under machine?

You can find it on the Display before you flash the Firmware or you have a look in your Fritzbox,  you can see the Ip under Wlan . It is the same IP before you update the Printer

Makkehu, 2025-03-14:
> How did you flash the original firmware? I've go the same hanging issue? I copied the original firmware to the folder on the USB but nothing happens when I reboot

It is very important that there are no files or folders on the USB drive except for a folder named aGVscF9zb3Nf. Inside this folder, there must be a file named update.swu. This is crucial; otherwise, the update process will never start.