utkabobr/DuckPro-Kobra3 #82: [question] install Debian and native klipper

Ziomal12, 2025-02-07:
Hello
I hope it's not too stupid of a question but is it possible to install normal klipper version on this printer? Especially now that you've got firmware flashing figured out how hard would it be to move it to native klipper? The printer config looks like would be accepted by klipper.

Please don't take it the wrong way. I really appreciate the work you're doing and understand that it took a great effort.

I'm looking into getting S1 combo and it would be no brainer if it run native klipper but right now, as someone coming from modded and klipperized ender 3 I can't imagine not being able to start a job using lan only wifi dashboard from orca or just in general be in full control.

On the other hand I'm worried that I'd lose come of the functionality (like bed leveling or filament changes).

ducati1198, 2025-02-07:
I understand your pain, because my too i can't wait for vanilla klippert. Spoolman doesn’t work good, i cant send pushover notifications and even a microcontroller (rpi pico) doesn’t work. Its sad that they didn’t realize full goklipper firmware

jbatonnet, 2025-02-07:
Hi @Ziomal12, if you ever get the S1 I would be interested in seeing how much it shares with the Kobra 3. I'm working on [Rinkhals](https://github.com/jbatonnet/Rinkhals), a custom firmware and I just added support for some Kobra 2 Pro printer as well as the Kobra 3. If the S1 is similar enough it would be pretty easy to add support as well.

I know that similar to DuckPro, my firmware doesn't offer a vanilla Klipper experience, but it covers basic local needs (printing from LAN, Orca, dashboard over WiFi and more) keeping all stock features intact.