utkabobr/DuckPro-Kobra3 #88: Error by upload and print file from Orca Sclicer V.2.3

AM6User26, 2025-02-17:
When I Upload a gcode directly from the Orca Slicer V2.3.0 on my Kobra 3 with the newest Duckpro version. I become follow Error.

error: typ = WebRequestError, code = 10011724, message = The gcode file does not contain the T command
01:23
File selected
01:23
File opened:flowrate__PLA_0.2_36m58s.gcode Size:719042

The Orca Slicer is new installed. Nothing change in the Printer Setup at Orca.

I upload the Gcode and the Moonraker file a this ticket.

P.S. Sorry for my Bad English.  

Nice greets from Germany.

[moonraker.log](https://github.com/user-attachments/files/18816963/moonraker.log)

[gcode.txt](https://github.com/user-attachments/files/18816980/gcode.txt)

AnTuAnObEr, 2025-02-24:
Hi,

I have the same problem with the same version of Orcaslicer: 
 - error: typ = WebRequestError, code = 10011724, message = The gcode file does not contain the T command.

Nice greets from France.

utkabobr, 2025-02-25:
See here: https://github.com/utkabobr/DuckPro-Kobra3/issues/11#issuecomment-2423197528