utkabobr/DuckPro-Kobra3 #89: Issue Reverting to Original Firmware on Anycubic Kobra 3

gkhnbjkxyz, 2025-02-17:
Hello,

I recently installed a third-party firmware on my Anycubic Kobra 3, but now I want to revert to the original firmware. However, I am encountering errors when trying to reinstall the stock firmware.

I have tried the following steps:

Downloaded the official firmware from Anycubic’s website.
Formatted my SD card to FAT32 and placed the firmware file on it.
Inserted the SD card and restarted the printer, but the firmware does not install.
Can you please guide me on how to properly revert to the original firmware? Do I need to use a different method, such as DF

Thank you for

Best regards,
Gökhan ÖZKAYNAK

rolandpierzga, 2025-02-17:
You need to format pendrive to FAT32, create folder an name it to aGVscF9zb3Nf. Copy original firmware inside this folder. Firmware need to be renamed to update.swu

gkhnbjkxyz, 2025-02-19:
Thank you for your help! Thanks to you, I was able to solve the problem.

closed 2025-02-19 by gkhnbjkxyz