utkabobr/DuckPro-Kobra3 #92: Run Duck Pro with the New Firmware 2.3.7 ?

Joerg1309, 2025-03-03:
Anycubic has released a New Firmware with Skip Objects and Flow Calibrations.

Can I use Duck Pro with this Firmware ? 
Can I Update the Printer when Duck Pro is on the Device ?

Is it possible to implement or use these features in Duckpro?

regards 
Smarty

Grossblom, 2025-03-03:
Would also like to know!

utkabobr, 2025-03-03:
> Latest duckpro release tested with 2.3.7, and no, incompatible. Gives 11407 error.

This error are mostly caused by printer.cfg incompatibility, I suppose it will be fixed with printer.cfg changes

christophervankammen, 2025-03-07:
Is anyone having any issues with Z axis sensor working when they upgrade to 2.3.7 then find its not compatible then try to get back to stock 2.3.5?

Art531, 2025-03-07:
> Is anyone having any issues with Z axis sensor working when they upgrade to 2.3.7 then find its not compatible then try to get back to stock 2.3.5?

No any issues with z after upgrade to 2.3.7 and after that to 2.3.7.1

MJFairbairn, 2025-03-07:
> Is anyone having any issues with Z axis sensor working when they upgrade to 2.3.7 then find its not compatible then try to get back to stock 2.3.5?

Yep tries to go straight through the bed if i go back to 2.3.5 tried several times but no success don't like 2.3.7.1 but stuck until they fix it, homes to back left, heats the nozzle to temp before bed e.t.c.

gelberlemmy, 2025-03-08:
Hello, is there anything new regarding 2.3.7 with Duck Pro?

quinman24, 2025-03-08:
yay i got an anycubic update when going to printer info when i was on duck pro. i updated and itr want back to stock firmware. tryed installing your new duck pro and got the 11407. installed old stock firmware, reinstalled duck pro and now fluidd and mainsail are not working. they open the page but can't connect to moonraker.

please help!?

killerplautze14, 2025-03-09:
hello utkabobr I have installed the new DuckPro 01.03.25 on my kobra 3 and now I can no longer access the mainsail, it always says If this message appears repeatedly, please have a look in the log file located at:~/printer_data/logs/moonraker.log I can no longer reset it to 2.3.5.3 what can I do

quinman24, 2025-03-09:
> hello utkabobr I have installed the new DuckPro 01.03.25 on my kobra 3 and now I can no longer access the mainsail, it always says If this message appears repeatedly, please have a look in the log file located at:~/printer_data/logs/moonraker.log I can no longer reset it to 2.3.5.3 what can I do

i noticed that the version that is labeled "latest" aka 01.03.25 is lower then 05.12.24 and usually versions go up not down i will test 05.12.24 and see if it works with 2.3.7.1 . ill get back to you guys if it works.

killerplautze14, 2025-03-09:
quinman24 I previously had the 05.12.24
then put the new one on 01.03.25 since then nothing has worked, not even resetting

utkabobr, 2025-03-09:
> i noticed that the version that is labeled "latest" aka 01.03.25 is lower then 05.12.24 and usually versions go up not down i will test 05.12.24 and see if it works with 2.3.7.1 . ill get back to [...]

It's not classical versions, they're release dates :)

utkabobr, 2025-03-09:
Btw updated printer.cfg to 2.3.7.1, should run fine now

killerplautze14, 2025-03-09:
hello utkabobr now it is stuck neither the duckpro nor the original firmware can be installed error 11407

MJFairbairn, 2025-03-09:
Hi when using Slicer Next while printing not showing print Temps Head Light Switch not working any more you can change everything ok and prints fine. Works ok through Fluidd and Mainsail perfectly.

killerplautze14, 2025-03-09:
Error fixed, works again for some reason I had 2 printer cfg on it, I managed to delete one now it works again :-)

killerplautze14, 2025-03-09:
new error

Moonraker Warnung

Supplied path (/userdata/app/gk/docs) for (docs) is invalid. Make sure
that the path exists and is not the file system root.

MJFairbairn, 2025-03-09:
Just create the docs dir in (userdata/app/gk) and it goes away.

killerplautze14, 2025-03-09:
another question how do I get the camera to work

Joerg1309, 2025-03-09:
> new error
> 
> Moonraker Warnung
> 
> Supplied path (/userdata/app/gk/docs) for (docs) is invalid. Make sure that the path exists and is not the file system root.

How I make this ?

Joerg1309, 2025-03-09:
> Just create the docs dir in (userdata/app/gk) and it goes away.

How I make this ?

killerplautze14, 2025-03-09:
the new duckpro has errors, it parks on the left and the second blob it makes on the print bed when starting a print

killerplautze14, 2025-03-09:
[image]
[image]
[image]
[image]

killerplautze14, 2025-03-09:
as you can see in the pictures the first blob is ok the second one is not at the end of a print it moves to the left cuts the filament pulls the filament back and stops

Joerg1309, 2025-03-09:
> as you can see in the pictures the first blob is ok the second one is not at the end of a print it moves to the left cuts the filament pulls the filament back and stops

I have the same Issue

MJFairbairn, 2025-03-10:
> another question how do I get the camera to work

Create the .camera file as usual it worked for me no problem.

Makkehu, 2025-03-11:
I don't recommend putting the latest duck pro firmware. There's a lot of change in printer.cfg. The cobra stopped working properly. Cleaning the nozzle does not start before calibrating the table. The thread is not cut, and the nozzle is not cleaned. Error 11407 is manifested if you set it to the original 2.3.7.1. In order to avoid errors, you need to install 2.3.7.1 firmware from the archive. I don't recommend this firmware to anyone at all, even though the work is done for which I'm grateful. But I can't figure out why I should change the algariticities of macros and printers. The beaver is not the same anymore.

utkabobr, 2025-03-12:
> I don't recommend putting the latest duck pro firmware. There's a lot of change in printer.cfg. The cobra stopped working properly. Cleaning the nozzle does not start before calibrating the table. [...]

Printer.cfg is upstreamed from base firmware. You can always revert back to the old one if you port changes needed to work.

Makkehu, 2025-03-14:
> У кого-нибудь возникали проблемы с работой датчика оси Z после обновления до версии 2.3.7, когда он переставал работать, а затем вы возвращались к версии 2.3.5?

Yes, this problem affects everyone. If you have once installed firmware 7.1 and then revert to 5.3, the Z-axis sensor will no longer work. This is due to the re-flashing of the extruder board. Unfortunately, this process is irreversible. We have tried many different solutions.

Makkehu, 2025-03-14:
> Printer.cfg is upstreamed from base firmware. You can always revert back to the old one if you port changes needed to work.

I made changes to printer.cfg, tried modifying many things, but all attempts were unsuccessful. For some reason, after installing DUCK Pro, the configuration file changes, but not significantly. If I restore the original printer.cfg, I get error 11407.

I conducted numerous experiments and even bought new extruder heads specifically for this, since the firmware overwrites them, causing problems when rolling back. So, I ruled out that issue. However, this does not help in getting the custom firmware to work correctly.

The most obvious errors are shown in the image above. During bed calibration, the nozzle does not heat up to 170°C (which is set by default, and in printer.cfg it is also 170°C). The extruder head immediately moves to the nozzle cleaning position while still cold.

But if you say that printer.cfg remains original, then why does it contain significantly modified data inside?

utkabobr, 2025-03-14:
> I made changes to printer.cfg, tried modifying many things, but all attempts were unsuccessful. For some reason, after installing DUCK Pro, the configuration file changes, but not significantly. If [...]

1. You can check gklib.log with logging level **debug**. You might be using old vanilla config if you get 11407
2. Printer.cfg is modified to disable wrong square corner velocity/input shaper config and some very small changes. Just diff with the vanilla one from 2.3.7.1
3. The firmware does NOT do any changes to mcu firmware (printhead and bed), that's why you should update to latest stock firmware before installing

Makkehu, 2025-03-14:
I certainly won't argue with you on the third point. Most likely, you just need some time, and you will see for yourself that the extruder head can be reflashed.

utkabobr, 2025-03-14:
> I certainly won't argue with you on the third point. Most likely, you just need some time, and you will see for yourself that the extruder head can be reflashed.

You definitely get another firmware on head mcu with new hardware. You can manually reflash it (Do not attempt it if you don't really sure) with instructions pinned in issues.
CFW installer NEVER attempts to reflash it for you, all the code is open-source, so anyone can confirm it.
Only stock firmware 2.3.7.1 can try to do something, I'm not changing parts that isn't necessary to run moonraker/mainsail