utkabobr/DuckPro-Kobra3 #95: Filament extrusion fails

gelberlemmy, 2025-03-10:
The print head moves to the ejector, extrudes and ejects it. Then it moves to the right to trigger the ejection mechanism. It then stops and extrudes onto the pressure plate. Printing then starts normally. Is there something wrong in the Printer.cfg ? I have already reset everything to the original. Does anyone have any tips? 
[image]

Makkehu, 2025-03-11:
It's the same problem. The person who made this firmware deliberately removed macros for kobra 3 combo. As a result, this firmware is not suitable for daily work. I genuinely don't understand why it's messing up something that worked well. And there is another problem - when auto-leveling the table, the printer stopped preheating before cleaning the nozzle. I don't recommend this shit at all.

Fichte79, 2025-03-20:
I have the Same Problem 

[image]

John-Disco, 2025-03-24:
The same problem